Limine-Bootloader/Limine #11: No efi system table field / support?

alula, 2020-05-29:
stivale only specifies UEFI boot flag but where's the system table?

Itay2805, 2020-05-29:
The flags tag is just meant for telling the kernel if it can call the bios after handover or not, the bootloader will call ExitBootServices by itself so passing the system table is quite useless. In theory we can pass the runtime services but they don't really offer much which is relevant to hobby osdev.

alula, 2020-05-29:
rtc clock/variables/firmware version might be quite useful in some cases imo

Itay2805, 2020-05-29:
Rtc clock is from the easiest to access, firmware version is probably as useless as cr8, variables are nice but really not necessary for hobby osdev. If there is something nice there is the reset system function since it has a shutdown but still not really worth it...

mintsuki, 2020-05-29:
Also, speaking about RTC, stivale is supposed to pass to the kernel a timestamp (in UNIX epoch format) of the time at boot. So reading RTC with UEFI is likely unneeded. stivale is supposed to provide, as much as possible, an agnostic platform for kernels; providing UEFI boot services goes outside the scope of stivale itself.

closed 2020-05-29 by mintsuki