Limine-Bootloader/Limine #44: Possibility of a limine-mkrescue type of utility?

Qix-, 2020-10-06:
During development, `grub-mkrescue` is pretty essential for building quick ISO images for testing in e.g. VMware or Qemu (via `-cdrom`).

Would it be possible to supply a `limine-mkrescue` utility that does something similar?

N00byEdge, 2020-10-06:
https://github.com/limine-bootloader/limine/blob/master/limine-install.c

Qix-, 2020-10-06:
@N00byEdge Yep, that installs the bootloader to a pre-existing device or  **raw** image format. It doesn't say anything about generating a standalone rescue ISO given a configuration file and a binary image.

N00byEdge, 2020-10-06:
Oh sorry, I thought that limine-install was what you were after and that you weren't aware of it. I personally don't see a point of iso images over raw ones, but I suppose that it could be something which has a use case.

Qix-, 2020-10-07:
Like I said, they're useful for development.

closed 2020-11-05 by mintsuki

Qix-, 2020-11-06:
@mintsuki @atiep I think you mis-tagged the issue number as the framebuffer changes in that commit have nothing to do with this issue.

mintsuki, 2020-11-09:
Yes, @AtieP wrongly mentioned this issue, sorry.

AtieP, 2020-11-09:
Sorry

Qix-, 2020-11-09:
No problem, can we get it reopened perhaps? :)

mintsuki, 2020-11-09:
If you want CD-ROM ISO support, you can perhaps open an issue for that specifically ;)

Qix-, 2020-11-09:
I'm confused, that's mostly what this issue is. Why isn't it being re-opened when it was falsley closed? GRUB has a `mkrescue` command to create quick ISO images, which is... literally exactly that, lol.