Limine-Bootloader/Limine #45: Feature request: Framebuffer's dimensions also specifiable in limine.cfg

rodri, 2020-10-24:
It might be useful that the dimensions of the framebuffer could be also specified in limine.cfg, per entry, like FRAMEBUFFER=1920x1080x32 or something.
It would only be used if the kernel requested a graphical framebuffer. Or even it would only be used if the request was a default request (0,0,0).
The rationale is that the automatically selected default dimensions might not be the desired ones, and it is easier to change limine.cfg, or to specify several entries with different dimensions and the same kernel, than to change the kernel's source code and recompile it, to adapt to different machines, for example.

AtieP, 2020-11-04:
To be honest I definitely agree with this one.

AtieP, 2020-11-04:
If mint agrees, I can try implementing this :smiley:

commit 168716a2 (2020-11-09): misc: Add config file options for specifying the resolution in a granular way. Other misc bug fixes. Fixes #45
closed 2020-11-09 by mintsuki