Limine-Bootloader/Limine #52: UEFI support

nexos-dev, 2020-11-11:
Hello,
BIOS has been superseded as of 2019. I think it would be a good idea to add a UEFI loader to the repo. I could contribute it in a series of PRs if you would like it.

Itay2805, 2020-11-13:
UEFI support in Limine is still under consideration and might be added in the future. Until then (#shameless plug) you could use [TomatBoot](https://github.com/TomatOrg/TomatBoot-UEFI) which is a stivale1/stivale2 compliant bootloader (for the most part), and it should be perfectly possible to have a drive with both limine and tomatboot at the same time.

nexos-dev, 2020-11-14:
Ok, thanks!

mintsuki, 2020-11-15:
We agreed that perhaps it is best if Limine stays BIOS-only since TomatBoot exists and it supports the same protocols as Limine does.

closed 2020-11-15 by mintsuki

nexos-dev, 2020-11-17:
Ok, makes sense. Thanks!

rizet, 2021-05-10:
TomatBoot is dead, seems Limine had to fill in it's shoes.

nexos-dev, 2021-06-10:
Agreed. That makes the most sense to me as it would be easier to just maintain Limine instead of Limine and TomatBoot.