Limine-Bootloader/Limine #53: Out of memory

PeterOSdev, 2020-11-14:
When trying to boot Linux with initrd I get:

PANIC: Out of memory

The same files loaded from a different drive, work.

mintsuki, 2020-11-14:
Can you please answer these questions:
- What version of Limine is the version in question?
- What filesystem(s) are the ones involved here?
- What is the drive/partition layout of the system?

And please, attach a copy of your configuration file. Thank you.

PeterOSdev, 2020-11-15:
It is Limine 0.6 release.
I did put Limine on an USB stick with a FAT32 partition (in which are the configuration

 file, vmlinuz and initrd).
My HD has two ext4 partitions. On the second partition are the same vmlinuz and initrd as on the stick).
I renamed the limine.cfg file before uploading because Github accepts only the TXT filename ending.
[limine.txt](https://github.com/limine-bootloader/limine/files/5542114/limine.txt)

PeterOSdev, 2020-11-15:
Addition, might be important: On the HD the booted files are symbolic links.

mintsuki, 2020-11-15:
That changes everything considering we do not support symlinks (yet). You can open a new issue for adding support for ext* symlinks.

That said it is true that this error isn't very clear; if you can run Limine from the unstable branch and add the `stage2.map` file (from the root of the repo) to your USB stick, then add a line that reads:
```
STAGE2_MAP=bios://:1/stage2.map
```
to your config file, then we'd be able to get a stacktrace to see what path control takes to get to the "Out of memory" error.

PeterOSdev, 2020-11-15:
Here the output of Limine attached as a picture
[image]
:
If it is too blurry I'll retry photographing it.

commit 6fd8eb05 (2020-11-16): ext2: Add initial symlink support. Fixes #53
commit be17ed18 (2020-11-16): ext2: Add initial symlink support. Fixes #53
closed 2020-11-16 by mintsuki