Limine-Bootloader/Limine #58: LLVM problems with kernel

nexos-dev, 2020-12-11:
Hello,
I am working on a kernel that uses stivale2 and compiles with LLVM and GNU. LLD, LLVM's linker, however, has a bug which prevents it from outputting the stivale2 header correctly and makes Limine bite the dust loading it. For this reason, it might be a good idea to have an alternative loading system in Stivale 2 that does what multiboot does (check in the first 8K or so for the image). That would make stivale2 work with kernels compiled with LLVM! I can contribute it if you would like.
Thanks,
nexos-dev

Itay2805, 2020-12-11:
There should be no problem using llvm with stivale2, and doing it how multiboot does is even more prone to error and would require a new section anyways (because you must have it in the first 8kb)

streaksu, 2020-12-11:
I would check for the attributes of the section, I myself experienced issues with LLD being weird with user-generated inside text.
The errors I found were fixed by explicitly setting the section attributes of the `.stivale2hdr` section containing the header by changing
```
.section .stivale2hdr
```
To
```
.section .stivale2hdr, "ax", @progbits
```
Please do modify it to suit your environment/assembler, the written code is for GAS, but all assemblers under the sun have similar directives.

nexos-dev, 2020-12-12:
I tried adding "aw" to the section and it worked! I'm probably not going to use LLVM anymore, as I ran into more problems with LLD.  Thanks for the help, though!

closed 2020-12-12 by nexos-dev