Limine-Bootloader/Limine #61: Limine not booting kernel

circutrider21, 2020-12-31:
The version of Limine that I used was Unstable, with commit hash 7f1dd08892e1e000284f05d5d87018dd45684b72

Limine will not boot any operating systems. Upon selecting the os that I want to boot, Limine spits out some information about Loading ELF before giving a blank screen and not booting the kernel nor doing anything.

I expected the kernel to start after selecting it on the menu screen.

Instead what happens is that it just gives me a blank screen.

All I did was run a vhd with a FAT-32 partition and limine installed (using limine-install) in qemu. Upon boot, the limine screen shows up, but when I select an os, it loads for a few seconds then gives me a black screen.

There was no stack trace at all. Neither was there a panic screen.

I have recreated this problem with 2 Operating Systems. First with [Skift OS](https://github.com/skiftOS/skift) Then with [This Tutorial](https://github.com/osdev-wiki/limine-barebones).

PS: I think the issue is with limine-install but I'm not yet sure.

mintsuki, 2020-12-31:
Hello, thank you for the bug report.

I have used the stivale-barebones repository you linked to test whether I could reproduce this issue, to no avail. It would be helpful to know more details about the host system alongside a more reproducible set up to try and recreate the issue.

It would be great if you could create a GitHub repository with all that is necessary for you to be able to reproduce the issue: Makefiles, limine config, kernel code... and post a link to it here, alongside info about your host OS like kernel, distro (where applicable), libc, and whatnot.

circutrider21, 2020-12-31:
Ok, I am working on it

circutrider21, 2021-01-01:
Fixed It! Apparently, it was a problem with qemu. I didn't allocate enough memory and by default, the qemu on my system would only allocate 8 MB. That is not enough for some reason. Anyways, allocating 256MB fixed the issue!

closed 2021-01-01 by circutrider21