Limine-Bootloader/Limine #72: 32-bit Stivale loader pushes 0 before boot info

micahswitzer, 2021-02-22:
**Limine version used**
Tested on the most recent release (v1.0.6)

**Describe the bug**
When the handoff from Limine to my kernel entry occurs, I pop off what I believe to be the address of the boot info struct, but instead I always get a null pointer (0).

**Expected behavior**
I expect to get a non-null address pointing to the boot info structure.

**Actual (broken) behaviour**
The popped value is always null.

**To Reproduce**
Checkout `187-Limine-Migration` from https://github.com/panix-os/Panix.git
```sh
git submodule update --init
make
make run
```

Observe boot info printed as null in the console.
```
Bootloader info at 0x0
```

**Additional context**
I believe that the bug is in `stage2/protos/stivale.c` in `stivale_spinup`.
I believe the code that reads
```c
"mov esp, dword ptr [esi]\n\t"
"push edi\n\t"
"push 0\n\t"
```
Should be
```c
"mov esp, dword ptr [esi]\n\t"
"push 0\n\t"
"push edi\n\t"
```
Since the spec says that first a null stack frame is pushed before the boot info is pushed. This change makes my code work properly.

If this indeed is an issue, I am willing to submit a PR to fix it.

jlxip, 2021-02-22:
The entry point in my kernel is a C++ function, and it works fine. I think, but I'm not sure (Mint, please confirm) that Limine targets this scenario, therefore pushes at the top of the stack the return address, just pop it if your entry point is in asm

Itay2805, 2021-02-22:
Yeah, it follows the stdcall abi, so your entry should follow it too

micahswitzer, 2021-02-22:
Ah right. That's how calling conventions work. I can't believe I got that mixed up. Thanks for the correction.

closed 2021-02-22 by micahswitzer

mintsuki, 2021-02-22:
I believe this is more of a wording issue in the specification.

Quoting it:
```
esp is set to the requested stack as per stivale header. An invalid return address of 0 is pushed to the stack before jumping to the kernel.

The physical address of the stivale structure (described below) is pushed onto this stack before the entry point is called.
```

Basically, the address to the structure is pushed first, followed by a return value of 0 and a `jmp`, as to emulate an x86 `call`. When the spec says "is pushed onto this stack before the entry point is called", that "called" refers to the `push 0` + code transfer operation.

In order for your kernel to obtain the proper value you'll have to read from `[esp + 4]` instead of `[esp]` as that would be your "return address" (of 0).

As other people mentioned, this is to make it possible for the entry point to be directly a C/ABI compliant function without the need for an assembly entry point.