Limine-Bootloader/Limine #80: Higher half 64bit kernel, different FIXED_HIGHER_HALF_OFFSET_64

LittleFox94, 2021-04-27:
**Limine version used**
Both binary release v2.3.6 and HEAD at the time of writing (3702060f695a248b29d807fe1ac69d99af0ab26e)

**Expected behavior**
I expected limine to load my 64bit stivale2 kernel image and jump into it.

**Actual (broken) behaviour**
It crashed with "stivale2: ELF64 load failure".

After a short debugging session and code crawl in limine, I figured this to be the problem: https://github.com/limine-bootloader/limine/blob/trunk/stage23/lib/elf.h#L8

Limine has another definition of "higher half" than my kernel, which has the following LOAD segments:
```
  LOAD           0x0000000000000000 0xffff800001000000 0xffff800001000000
                 0x0000000000015bd6 0x0000000000015bd6  R      0x1000
  LOAD           0x0000000000015be0 0xffff800001016be0 0xffff800001016be0
                 0x000000000000c5d2 0x000000000000c5d2  R E    0x1000
  LOAD           0x00000000000221c0 0xffff8000010241c0 0xffff8000010241c0
                 0x0000000000000068 0x00000000005322b8  RW     0x1000
```

I believe 0xffff8... to be the correct start of the higher half on 4-level amd64, using this value also works fine when changing the line linked above and with my own loader.

**To Reproduce**
Build a 64 bit stivale2 kernel image with `-Wl,--image-base,0xFFFF800001000000` (might be clang/ld.lld specific flag)

**Stacktrace**
Will happily provide if needed, but I think there is enough info in here.

**Additional context**
With limine being a generic bootloader, I'd expect this to be at least configurable somehow. Maybe it isn't needed at all, loading the kernel at a random place while using the image in limine and using whatever is supplied as VADDR for the final mapping.

Anyway, great project and I really like the stivale2 protocol, from what I've read so far - so kudos for that :)

chocabloc, 2021-04-27:
Higher half kernels are usually loaded at `0xffffffff80000000`, so it's a totally reasonable assumption and not 'broken' imo. But yeah, the ability to load at arbitrary addresses would be nice

LittleFox94, 2021-04-27:
Maybe I didn't read a spec yet, but for me higher half was always anything with the MSB set. But definition of that probably isn't important here.

I'd try to build a patch that allows loading the unchanged barebones samples and my kernel, if there is a higher than 0 chance to have this merged after all ^^

mintsuki, 2021-04-27:
Quoting the stivale2 specification: 
> stivale2 natively supports (only for 64-bit kernels) and encourages higher half kernels. The kernel can load itself at 0xffffffff80000000 + phys_base (as defined in the linker script) and the bootloader will take care of everything else, no AT linker script directives needed.

So it is clear and not a bug that the *kernel's* higher half base starts at `0xffffffff80000000` and not `0xffff800000000000`.

As to why that is the case, most compilers are able to optimise better if code is situated at either the bottom 2GiB of address space, or the top 2GiB. Top 2GiB starting at `0xffffffff80000000`.

Compilers like GCC or clang can do optimisations for kernels loaded at `top - 2GIB` by means of the `-mcmodel=kernel` flag.

Loading anywhere else in the address space but the top or bottom 2GiB will require the use of `-mcmodel=large` which incurs in some optimisation deficits (this is for when it comes to non-relocatable kernels).

Another reason why 2GiB from top is a more desirable load location for the kernel image is that `0xffff800000000000` is *not* a fixed higher half start, and it is bound to change depending on whether 4 or 5-level paging (or anything else added in the future) is used. For example for 5-level paging, the higher half would start at `0xff00000000000000` instead.

LittleFox94, 2021-04-27:
Thanks @mintsuki, some very interesting info in there I didn't know yet :)

Currently I have mcmodel=large, so I'd probably benefit from changing the address.

I'll tackle to patch this (as enhancement, ACK it's not a bug) anyway for fun, if you don't tell me it won't be merged anyway x)

mintsuki, 2021-04-27:
If the patch to allow kernels to be loaded at arbitrary higher half addresses is clean enough and doesn't break compatibility, I won't say no :)

LittleFox94, 2021-04-27:
Oki :)

Any other things I'll have to check besides the unchanged examples still loading and my (bad base address kernel) loading?

mintsuki, 2021-04-27:
The [stivale](https://github.com/stivale/stivale) spec would have to be modified accordingly to reflect the new behaviour. An additional PR there first would do.

Once I take a look at that and it's okay to merge there, send the PR for Limine itself and I'll try to triple check if everything still works before merging.

LittleFox94, 2021-04-27:
Ok :)

Can not promise it happening any time soon or at all, but will very probably look at it :)

closed 2021-05-28 by mintsuki

Qix-, 2023-04-21:
> Another reason why 2GiB from top is a more desirable load location for the kernel image is that `0xffff800000000000` is not a fixed higher half start, and it is bound to change depending on whether [...]

FWIW anything with the MSB set would be higher half and thus fall into the requirement of "higher half". This is further enforced by x86_64 systems since bit 47 (or whatever MSB for the address space) is sign-extended, and on ARM/MIPS due to the MSB being checked explicitly by the arch, if I understand correctly. The requirement that virtual addresses sign extend the last bit is explicitly for future-proofing the higher-half design, for the specific reason you stated - the levels of paging might increase over time.

While the part about optimizations is certainly interesting, I too was confused as to why higher-half didn't _actually_ mean higher-half in the protocol docs, and this was the only place I could find reasoning as to why.