Limine-Bootloader/Limine #106: limine: failed to allocate memory ranges

rizet, 2021-08-29:
**Limine version used**
*2.49.1*

**Expected behavior**
*Boot without Limine errors.*

**Actual (broken) behaviour**
*Failed to allocate memory ranges.*

**To Reproduce**
Clone and run the [Skylight](https://github.com/rizet/skylight) repo.

**Stacktrace**
[image]

**Additional context**
*Occurred after copypasting a linked-list heap implementation and adding some modifications.
(I do not expect my OS to fully function in it's current state, but I do expect it to boot.)
I have resumed this project after a six-week hiatus and the previously used version of Limine was v2.17 (it auto-updates).*

mintsuki, 2021-08-29:
Let me premise this by saying that having a 256MiB large load segment in an executable isn't the most normal thing ever.

That said I "fixed" this in 12566c5.

closed 2021-08-29 by mintsuki

rizet, 2021-08-29:
Having a 256MiB large load segment was not my intention.

With that being said I do believe it is important to be able to load large segments... as long as you have enough RAM and it's less than 2GB.