Limine-Bootloader/Limine #133: Section .stivale2hdr not found

TheAirBlow, 2021-11-17:
Code can be viewed [here](https://github.com/corruptTeam/corruptKernel)
I use example's `kernel.ld` and a similar `Makefile`. 
The only difference are presence of symbols and `asm`.

Here is what `objdump` says:
```
kernel/bin/kernel.elf:     file format elf64-x86-64
kernel/bin/kernel.elf
architecture: i386:x86-64, flags 0x00000112:
EXEC_P, HAS_SYMS, D_PAGED
start address 0xffffffff80000050

Program Header:
    NULL off    0x0000000000000000 vaddr 0x0000000000000000 paddr 0x0000000000000000 align 2**3
         filesz 0x0000000000000000 memsz 0x0000000000000000 flags ---
    LOAD off    0x0000000000001000 vaddr 0xffffffff80000000 paddr 0xffffffff80000000 align 2**12
         filesz 0x0000000000000285 memsz 0x0000000000000285 flags r-x
    LOAD off    0x0000000000001285 vaddr 0xffffffff80001285 paddr 0xffffffff80001285 align 2**12
         filesz 0x000000000000017b memsz 0x000000000000017b flags r--
    LOAD off    0x0000000000000400 vaddr 0xffffffff80002400 paddr 0xffffffff80002400 align 2**12
         filesz 0x0000000000000000 memsz 0x0000000000000010 flags rw-

Sections:
Idx Name          Size      VMA               LMA               File off  Algn
  0 .text         00000285  ffffffff80000000  ffffffff80000000  00001000  2**0
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 .rodata       0000000d  ffffffff80001285  ffffffff80001285  00001285  2**0
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  2 .eh_frame     00000138  ffffffff80001298  ffffffff80001298  00001298  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  3 .note.gnu.property 00000030  ffffffff800013d0  ffffffff800013d0  000013d0  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  4 .bss          00000010  ffffffff80002400  ffffffff80002400  00001400  2**3
                  ALLOC
  5 .debug_info   0000075f  0000000000000000  0000000000000000  00001400  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
  6 .debug_abbrev 0000030d  0000000000000000  0000000000000000  00001b5f  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
  7 .debug_aranges 00000090  0000000000000000  0000000000000000  00001e6c  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
  8 .debug_line   0000022d  0000000000000000  0000000000000000  00001efc  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
  9 .debug_str    00000366  0000000000000000  0000000000000000  00002129  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 10 .debug_line_str 000000c4  0000000000000000  0000000000000000  0000248f  2**0
                  CONTENTS, READONLY, DEBUGGING, OCTETS
 11 .comment      00000012  0000000000000000  0000000000000000  00002553  2**0
                  CONTENTS, READONLY
SYMBOL TABLE:
0000000000000000 l    df *ABS*  0000000000000000 kernel.c
0000000000000000 l    df *ABS*  0000000000000000 mem.c
0000000000000000 l    df *ABS*  0000000000000000 gop.c
ffffffff80000000 g     F .text  0000000000000050 stivale2_get_tag
ffffffff8000013a g     F .text  0000000000000024 gop_init
ffffffff800000ca g     F .text  000000000000003e memcpy
ffffffff80000215 g     F .text  000000000000003d draw_pixel
ffffffff800001be g     F .text  0000000000000057 clear
ffffffff80000050 g     F .text  000000000000007a _start
ffffffff80002408 g     O .bss   0000000000000008 cursorpos
ffffffff80002400 g     O .bss   0000000000000008 framebuffer
ffffffff80000108 g     F .text  0000000000000032 memset
ffffffff8000015e g     F .text  0000000000000060 draw_filled_rectangle
ffffffff80000252 g     F .text  0000000000000033 get_pixel
```

`PHDRS` header is present, but `.stivale2hdr` is not.
Any thoughts on this?

mintsuki, 2021-11-17:
Hello.  When it comes to the issue of .stivale2hdr missing, that section will only be created in the executable if any object file of the project places something inside it. From what I see, this is done nowhere in the project. An example of how to do this can be found here (https://github.com/limine-bootloader/limine-barebones/blob/master/stivale2-barebones/kernel/kernel.c#L48-L76).

The latter error you show is simply due to you not having the  program installed. (The error message is incredibly clear about this).

Hope this helps.

TheAirBlow, 2021-11-17:
Sorry for being such a noob.
Forgot that this structure was required.

closed 2021-11-17 by TheAirBlow

mintsuki, 2021-11-17:
PS I am a disaster at using GH cli lmao

TheAirBlow, 2021-11-17:
bruh

mintsuki, 2021-11-17:
(I should have fixed the mess I made, GH should really offer an option to revert accidental message edits)