Limine-Bootloader/Limine #134: Terminal not printing anything

nailuj29gaming, 2021-11-17:
**Limine version used**
2.64

**Expected behavior**
The terminal prints

**Actual (broken) behaviour**
My kernel just shows a blank screen

**To Reproduce**
code can be found at https://github.com/nailuj29gaming/esos

**Additional context**
I'm using the zigvale library to generate the stivale2 header, but I don't think the issue is there

mintsuki, 2021-11-17:
I don't know zig, but you don't seem to specify the identifier [here?](https://github.com/nailuj29gaming/esos/blob/main/kernel/src/stivale/tags.zig#L3-L9)

RiscInside, 2021-11-17:
Can you check if the terminal tag gets supplied or not?

You can do that, for example, by removing `while (true)` hang loop and replacing it with division by 0

mintsuki, 2021-11-17:
Upon further examination (I built the project and tested it) - The issue is caused by the fact that SSE instructions are emitted by the compiler. This is obviously bogus since you need to enable SSE yourself if you want to use it (and usually kernels disable SSE for this, and other reasons).

There is another issue that masked this problem: When running qemu with `-M q35`, instead of crashing/resetting on triple faults, the machine just hangs, masking the issue. Running qemu without that flag, and with `-d int -no-reboot -no-shutdown -M smm=off` quickly revealed the problem.

closed 2021-11-17 by mintsuki