Limine-Bootloader/Limine #137: Loading Linux kernel panics early

jinks, 2021-11-26:
**Limine version used**
2.67.2

**Expected behavior**
Boots the Linux kernel.

**Actual (broken) behaviour**
PANIC: linux: Unable to set video mode

**To Reproduce**
Use the following config
```
TIMEOUT=5

:Arch Linux

PROTOCOL=linux
KERNEL_PATH=boot:///vmlinuz-linux
CMDLINE=root=/dev/vda2 rw loglevel=3 quiet
MODULE_PATH=boot:///initramfs-linux.img
```

**Stacktrace**
```
Stacktrace:
  [0xb63a] <panic+0x4d>
  [0x1a85a] <linux_load+0x3ba>
  [0x1050c] <stage3_addr+0x2dc>
  [0x936d] <entry+0x133>
End of trace. Press CTRL+ALT+DEL to reboot.
```

**Additional information**
QEMU VM / VPS accessed via noVNC.

jinks, 2021-11-26:
Adding `RESOLUTION=640x480x16` to the entry fixed it.

There was no indication what resolutions were available, using a 32bpp mode (eider adding x32 or omitting it) resulted in the above PANIC.

mintsuki, 2021-11-26:
This is my proposed alternative change. (dab74fd)

Try to see if with this you manage to boot without specifying the resolution.

mintsuki, 2021-11-26:
And this one (6b0c9c4) should, in your specific case, avoid the need to fall back all the way to text mode.

mintsuki, 2021-11-27:
Fixed in 2.67.3

closed 2021-11-27 by mintsuki