Limine-Bootloader/Limine #141: MBR on EXT4: Blank screen after loading message?

SomethingGeneric, 2021-12-06:
**Limine version used**
2.68.2 (binary branch)

**Expected behavior**
BusyBox init to load and thus bring up the system

**Actual (broken) behaviour**
Blank screen? I think the system is up since there's CPU usage in the VM

**To Reproduce**
Try booting `./build/ikeda` after running `./build.sh qemu` from https://git.tar.black/ikeda/Ikeda/-/tree/limine (ensure `limine` branch :)  )
Expected behavior can be seen doing the same with the `main` branch (using GRUB2)

**Stacktrace**
N/A

**Additional context**
I saw another closed issue about VM memory, so I tried using virt-manager instead of just pure qemu

`/boot/limine.cfg` contents:
```
PROTOCOL=linux
KERNEL_PATH=boot:///boot/bzImage
KERNEL_CMDLINE=quiet
``` (same behavior w/o `quiet` param.)

I'm happy to share the source disk image(s) in question.

SomethingGeneric, 2021-12-06:
(edited to fix formatting goof)

leapofazzam123, 2021-12-06:
Looks like it also happened when loading linux kernels in other file systems like ISO 9660

leapofazzam123, 2021-12-06:
Can you try to get the output of `-serial stdio` and `-debugcon stdio` in QEMU?

mintsuki, 2021-12-06:
I cannot reproduce this by simply attempting to load a Linux kernel from an ext4 partition.
Please send over the image used to reproduce this by whatever means you want (may I suggest maybe Google Drive?)

mintsuki, 2021-12-06:
> Looks like it also happened when loading linux kernels in other file systems like ISO 9660

I have no idea what bug is being referenced here, but if you have some reproducible example, please share it, also keep it on topic.

> Can you try to get the output of `-serial stdio` and `-debugcon stdio` in QEMU?

That will not work as the prebuilt image is not compiler with E9 output enabled.

SomethingGeneric, 2021-12-06:
@mintsuki here's `ikeda-limine` as a compressed xz of the raw image https://drive.google.com/file/d/1JcntEsKLQUERKqbj6xzno5LLmkoRWWAf/view?usp=sharing

here's `ikeda-grub` as a compressed xz of the image
https://drive.google.com/file/d/1oPehgNVody1OKp1pfhwcs9F8BydlU4D_/view?usp=sharing

mintsuki, 2021-12-06:
Your kernel doesn't appear to support a framebuffer console which is what Limine defaults to. As per config documentation, adding `TEXTMODE=yes` reveals the kernel's output, which then reveals a panic due to the root not having been specified.

Specifying the root in the kernel's command line equalises the outputs of GRUB and Limine.

SomethingGeneric, 2021-12-06:
ah, ok. I'm new to configuring kernels so I guess I missed that. Thanks for the help!

closed 2021-12-06 by SomethingGeneric

mintsuki, 2021-12-06:
You're welcome!

leapofazzam123, 2021-12-07:
> I have no idea what bug is being referenced here, but if you have some reproducible example, please share it, also keep it on topic.

https://www.dropbox.com/s/hflmfvjbp8610ql/test.iso?dl=0

leapofazzam123, 2021-12-07:
In UEFI, the kernel boots fine (and also boots to fbcon), meanwhile in BIOS, it's just stuck at loading kernel

mintsuki, 2021-12-07:
> In UEFI, the kernel boots fine (and also boots to fbcon), meanwhile in BIOS, it's just stuck at loading kernel

This wasn't a technical bug per se but just an issue where if an initramfs was gz compressed, it would not show the loading message while decompressing it (fixed in 2be3776).

Now, the image works for me, but you have to keep in mind that gz decompression with tinf is slow, and especially so if not using hardware virtualisation, so it wasn't hanging and there was no issue with ISO or anything else, it was just decompressing without showing a loading message and taking a long time because you were not using hardware virtualisation.

I recommend not using gz compression unless you really need to save space on disk, as the speed tradeoff is usually not worth it.