Limine-Bootloader/Limine #144: Failed to open kernel with path 'boot:///vmlinuz-linux'. Is the path correct?

fiatjaf, 2021-12-25:
**Limine version used**
Limine v2.73.4

**Expected behavior**
I expected Linux to boot.

**Actual (broken) behaviour**
`PANIC: linux: Failed to open kernel with path 'boot:///vmlinuz-linux'. Is the path correct?`

**To Reproduce**
Install Arch Linux, follow the default install instructions and then these: https://wiki.archlinux.org/title/Limine

**Stacktrace**
```
Stacktrace:
  [0x9cc8] <panic+0x9d>
  [0x1d31f] <multiboot1_load+0x3f>
  [0x1a4ff] <boot+0x12f>
  [0x1b1a3] <_menu+0xbb3>
```

**Additional context**

Partitions:
- `/dev/sda1` -> swap
- `/dev/sda2` -> root

The `/boot` directory contains a file `vmlinuz-linux`, `initramfs...` and also a `limine.cfg`:

```
TIMEOUT=5

:Arch Linux

PROTOCOL=linux
KERNEL_PATH=boot:///vmlinuz-linux
CMDLINE=root=UUID=<uuid> rw loglevel=3 quiet
MODULE_PATH=boot:///initramfs-linux.img
```

fiatjaf, 2021-12-25:
Fixed by changing the paths to `boot:///boot/...` on `limine.cfg`, it seems that the instructions at https://wiki.archlinux.org/title/Limine are wrong or outdated.

closed 2021-12-25 by fiatjaf

StaticSaga, 2021-12-25:
The instructions aren't wrong, it does depend on your setup. boot:/// is the partition where limine is, which /boot on UEFI systems and generally / on BIOS systems.
> Note: In a Limine config, boot:/// represents the partition on which limine.cfg is located. In case there is no separate /boot partition, and limine.cfg resides on the root partition instead, then, in the following example, boot:/// should instead be boot:///boot/.