Limine-Bootloader/Limine #150: Limine will fall into panic while booting on a hyper-v virtual machine

jjwang, 2022-01-26:
Limine will fall into panic while booting on a hyper-v virtual machine. I attached screenshot as below:

[image]

Andy-Python-Programmer, 2022-01-27:
I am not able to reproduce it. Can you provide the settings for your hyper-v machine?

jjwang, 2022-01-27:
I tested with limine-barebones project without too much modifications. The limine version is 2.24. I also tested the latest binary and panic also happened. The hyper-v machine is default gen-1 setting and I launched limine from cdrom. I hope it can help.

mintsuki, 2022-01-28:
Please, try using the latest Limine release and see if you can reproduce this issue.

jjwang, 2022-02-01:
Thanks for your suggestion! I will retry with lastest release and report later.

jjwang, 2022-02-07:
I tried to build with latest limine code and stivale2-barebones, the panic changes to:

[image]

czapek1337, 2022-02-07:
Can somewhat reproduce on default Hyper-V gen1 VM using stivale2-barebones with Limine v2.83, although I'm getting a different panic.

[image]

czapek1337, 2022-02-07:
I also noticed that if you run the VM with at least 1946MiB of memory assigned to it you instead get a different panic 🤔

[image]

commit b80fb367 (2022-02-09): misc: Misc Hyper-V related fixes. Closes #150
commit f2b468c4 (2022-02-10): misc: Misc Hyper-V related fixes. Closes #150
closed 2022-02-09 by mintsuki