Limine-Bootloader/Limine #160: Issues when building with LLVM binutils + lld

Th3T3chn0G1t, 2022-03-07:
**Limine version used**
trunk@be2b099b

**Expected behaviour**
Limine would build with full LLVM toolchain

**Actual (broken) behaviour**
The `configure` script rejects `lld` (`ld.lld`) as it's version string contains "compatible with GNU linkers" not "Free Software Foundation".
It also rejects `llvm-ar`, `llvm-as`, `llvm-objdump`, `llvm-objcopy` and `llvm-readelf`, which don't contain any reference to GNU in their version strings.

[Patching out those checks](https://pastebin.com/LxDXk9Bz) - the build progresses until it attempts to link `common-uefi32/limine_nomap.elf` ([log](https://pastebin.com/xABTkfkU)).

**To Reproduce**
Configuring with `./configure CC=clang LIMINE_LD=ld.lld LIMINE_AR=llvm-ar LIMINE_AS=llvm-as LIMINE_OBJDUMP=llvm-objdump LIMINE_OBJCOPY=llvm-objcopy LIMINE_READELF=llvm-readelf` (Yes I know `TOOLCHAIN` exists I was just going through them one by one). I have tried on macOS and on Linux with both having the same behaviour.

**Stacktrace**
N/A

**Additional context**
I am using LLVM 13.0.1 (from Homebrew on macOS, from source on Linux).

Th3T3chn0G1t, 2022-03-07:
The `relro` bug seems to be the same one `systemd-boot` has [here](https://github.com/systemd/systemd/issues/13800)

mintsuki, 2022-03-07:
I am aware, the configure script *intentionally* rejects LLVM binutils as GNU binutils is a build dependency of Limine. This is clearly stated in the main readme file as well.

closed 2022-03-07 by mintsuki