Limine-Bootloader/Limine #162: PANIC: stivale2: Cannot use text mode with UEFI

ethindp, 2022-03-13:
**Limine version used**
2.88 (latest binary release)

**Expected behaviour**
My kernel should boot fine.

**Actual (broken) behaviour**
Limine believes I'm requesting text mode even though I'm not.

**To Reproduce**
Not honestly sure what I did wrong, here's how I declared everything:

* Stack: pointed to kernel stack (Limine still thinks stack is 0x0)
* Flags: `(1<<1)|(1<<2)|(1<<3)|(1<<4)`
* Tags:
    * Any video tag, preference 0
    * FB tag, BPP 32
    * Terminal tag
    * SMP tag, use X2APIC
    * 5-level paging tag
    * Unmap NULL tag

**Stacktrace**
```
stivale2: Loading kernel `boot:///EFI/BOOT/kernel`...
stivale2: Physical base: 0x7856e000
stivale2: Virtual base:  0xffffffff80000000
stivale2: 64-bit kernel detected
stivale2: Kernel slide: 0x0
stivale2: Entry point at: 0xffffffff80000420
stivale2: Requested stack at: 0x0
acpi: Found RSDP at 0x7fb7e000
acpi: Found XSDP at 0x7fb7e014
acpi: Found SMBIOS 32-bit entry point at 0x7f9ab000
stivale2: Current epoch: 1647183989
PANIC: stivale2: Cannot use text mode with UEFI.
Stacktrace:
  [0x7daa3a81] <panic+0x81>
  [0x7da8a15b] <stivale2_load+0x1d9b>
  [0x7da92dd2] <boot+0x1a2>
  [0x7da94541] <_menu+0x16e1>
End of trace. Press a key to return to menu.
```

**Additional context**
I am not requesting text mode. Booting in UEFI. Configuration file follows:

```
TIMEOUT=0
DEFAULT_ENTRY=1
GRAPHICS=yes
MENU_FONT_SCALE=2x2
EDITOR_ENABLED=no
VERBOSE=yes
SERIAL=yes
RANDOMIZE_MEMORY=yes
:Kernel
PROTOCOL=stivale2
KASLR=yes
KERNEL_PATH=boot:///EFI/BOOT/kernel
```

mintsuki, 2022-03-13:
I cannot reproduce this. Are you sure all the header tags have been properly linked to each other?

ethindp, 2022-03-13:
I think so... Here are my tag structures:

```rust
#[repr(C, align(4))]
struct Stack([u8; MAX_STACK_SIZE]);
static STACK: Stack = Stack([0; MAX_STACK_SIZE]);

// Tags
static VIDEO_TAG: StivaleAnyVideoTag = StivaleAnyVideoTag::new()
    .preference(0)
    .next(&FB_TAG as *const StivaleFramebufferHeaderTag as *const ());
static FB_TAG: StivaleFramebufferHeaderTag = StivaleFramebufferHeaderTag::new()
    .framebuffer_bpp(32)
    .next(&TERMINAL_TAG as *const StivaleTerminalHeaderTag as *const ());
static TERMINAL_TAG: StivaleTerminalHeaderTag =
    StivaleTerminalHeaderTag::new().next(&SMP_TAG as *const StivaleSmpHeaderTag as *const ());
static SMP_TAG: StivaleSmpHeaderTag = StivaleSmpHeaderTag::new()
    .flags(StivaleSmpHeaderTagFlags::X2APIC)
    .next(&LVL5_PG_TAG as *const Stivale5LevelPagingHeaderTag as *const ());
static LVL5_PG_TAG: Stivale5LevelPagingHeaderTag = Stivale5LevelPagingHeaderTag::new()
    .next(&UNMAP_NULL_TAG as *const StivaleUnmapNullHeaderTag as *const ());
static UNMAP_NULL_TAG: StivaleUnmapNullHeaderTag = StivaleUnmapNullHeaderTag::new();

#[link_section = ".stivale2hdr"]
#[used]
static BOOT_LOADER_HEADER: StivaleHeader = StivaleHeader::new()
    .stack(STACK.0.as_ptr_range().end)
    .flags((1 << 1) | (1 << 2) | (1 << 3) | (1 << 4))
    .tags(&VIDEO_TAG as *const StivaleAnyVideoTag as *const ());
```

ethindp, 2022-03-14:
I've attached my kernel binary; I'm not sure what other information I can give you. The only thing I did to cause this error was upgrading Limine.
[kernel.gz](https://github.com/limine-bootloader/limine/files/8248049/kernel.gz)

mintsuki, 2022-03-14:
Limine cannot find any tag, apparently. Are you setting the tag IDs correctly?

ethindp, 2022-03-14:
Yeah... It was working fine before I updated to 2.88...

mintsuki, 2022-03-14:
Try adding a `PT_DYNAMIC` PHDR to your linker script. See the linker script at https://github.com/stivale/stivale2-barebones.

This should've already been present in your linker script. Limine prior to 2.88 used the section table in order to find the `.dynamic` section but this is in violation of the ELF specification and was thus changed. Kernels which were properly linked prior to this change will experience no difference but your kernel lacks a `PT_DYNAMIC` PHDR, so it does.

ethindp, 2022-03-14:
I'm not quite sure why this is the case. My linker script is:

```ld
OUTPUT_FORMAT(elf64-x86-64)
OUTPUT_ARCH(i386:x86-64)

ENTRY(_start)

/* Define the program headers we want so the bootloader gives us the right */
/* MMU permissions */
PHDRS
{
    null    PT_NULL    FLAGS(0);
    text    PT_LOAD    FLAGS(0x5);
    rodata  PT_LOAD    FLAGS(0x4);
    data    PT_LOAD    FLAGS(0x6);
}

SECTIONS
{
    /* We wanna be placed in the topmost 2GiB of the address space, for optimizations */
    /* and because that is what the stivale2 spec mandates. */
    . = 0xffffffff80000000;

    .text : {
        *(.text .text.*)
    } :text

    /* Move to the next memory page for .rodata */
    . += CONSTANT(MAXPAGESIZE);

    /* We place the .stivale2hdr section containing the header in its own section, */
    /* and we use the KEEP directive on it to make sure it doesn't get discarded. */
    .stivale2hdr : {
        KEEP(*(.stivale2hdr))
    } :rodata

    .rodata : {
        *(.rodata .rodata.*)
    } :rodata

    /* Move to the next memory page for .data */
    . += CONSTANT(MAXPAGESIZE);

    .data : {
        *(.data .data.*)
    } :data

    .bss : {
        *(COMMON)
        *(.bss .bss.*)
    } :data
}
```

And that looks nearly identical to the one you linked to. I'm compiling with Rusts `x86_64-unknown-none` target and giving it my linker script as an extra argument.

Andy-Python-Programmer, 2022-03-14:
@ethindp try and use this target json file instead of using the `x86_64-unknown-none` target:

```json
{
  "arch": "x86_64",
  "code-model": "kernel",
  "cpu": "x86-64",
  "data-layout": "e-m:e-p270:32:32-p271:32:32-p272:64:64-i64:64-f80:128-n8:16:32:64-S128",
  "disable-redzone": true,
  "executables": true,
  "features": "-mmx,-sse,+soft-float",
  "has-rpath": false,
  "has-thread-local": false,
  "linker-flavor": "ld.lld",
  "linker": "ld.lld",
  "llvm-target": "x86_64-unknown-none",
  "max-atomic-width": 64,
  "os": "none",
  "panic-strategy": "abort",
  "pre-link-args": {
    "ld.lld": [
      "-z",
      "max-page-size=0x1000",
      "-T",
      "kernel/src/arch/x86_64/x86_64-bare.ld"
    ]
  },
  "stack-probes": {
    "kind": "call"
  },
  "target-c-int-width": "32",
  "target-endian": "little",
  "target-pointer-width": "64"
}
```

ethindp, 2022-03-15:
Thanks, that worked!

ethindp, 2022-03-15:
I should clarify: the linker script modification worked. I don't mind using a built-in rust target TBH.

closed 2022-03-15 by ethindp