Limine-Bootloader/Limine #165: Example code in `test` folder is broken

Andrej123456789, 2022-04-05:
**Limine version used**
3.0

**Expected behaviour**
To see printed text, framebuffer, memory, acpi etc

**Actual (broken) behaviour**
Watch video running both in Qemu and Virtual Box.

https://user-images.githubusercontent.com/83548580/161832756-21b9cc2b-a0ad-4a74-b36f-12c4b6691ef6.mov

**To Reproduce**
Copied code from `test` folder.

**Stacktrace**
[VBox.log](https://github.com/limine-bootloader/limine/files/8421256/VBox.log)
Vbox.log file.

mintsuki, 2022-04-05:
This is obviously *not* the test image and I *did* test the test image on virtualbox and it works fine.

Your vbox log shows pretty clearly that you're page faulting on a RIP fetch inside your kernel (or at the very least well past Limine's control), so, wild guess, you're probably executing garbage past the end of a function or calling some invalid address.

mintsuki, 2022-04-05:
On further inspection, it seems suspicious that in your log there is clearly randomisation of the HHDM address but your RIP appears non-randomised. If you want further help you'll have to share your source code.

closed 2022-04-05 by mintsuki

Andrej123456789, 2022-04-05:
https://github.com/ringwormGO-organization/ringOS/blob/Limine2/kernel/kernel.c
Here is code

mintsuki, 2022-04-05:
So: you took the code from the Limine test (including the linker script), but then you took the makefile from the barebones example. The barebones example makes a non-PIE kernel while the Limine test does.

Use the linker script from the barebones example, so you're not attempting to use a linker script meant for a PIE kernel when you're compiling objects as no-PIC.

Andrej123456789, 2022-04-06:
Ok. I changed linker script but this time Virtual Box works but nothing printing

EnderIce2, 2022-04-06:
> On further inspection, it seems suspicious that in your log there is clearly randomisation of the HHDM address but your RIP appears non-randomised. If you want further help you'll have to share your source code.

https://github.com/EnderIce2/FennixProject