Limine-Bootloader/Limine #167: KASLR not performed using Limine protocol

HildarTheDorf, 2022-04-07:
**Limine version used**
trunk bcced191a4d2a45a8063d40a78d5a40ada9cd1a0

**Expected behaviour**
KASLR performed for PIE kernels, virtual and physical address relocated

**Actual (broken) behaviour**
Kernel is always mapped at P: 0x0000000007AF3000 V: 0xFFFFFFFF80000000 when booting with the Limine protocol.

**To Reproduce**
Comple kernel with the following flags:
COMFLAGS=-mno-red-zone -O2 -mno-sse -mno-mmx -mno-80387 -Wall -Wextra -pedantic
CFLAGS=${COMFLAGS} -std=c17 -Iinclude -ffreestanding -fno-asynchronous-unwind-tables -fPIE -pipe
LDFLAGS=${COMFLAGS} -Wl,-pie -nostdlib -z max-page-size=4096

Observe non randomized values in limine_kernel_address_response

**Stacktrace**
N/A

**Additional context**
Booting with stivale2 results in correctly randomized physical and virtual addresses.

HildarTheDorf, 2022-04-07:
Virtual randomization performed when adding a PT_DYNAMIC segment to linker script, which is okay but different to when booting with stivale2.
Physical randomization is still not performed.

mintsuki, 2022-04-12:
Hello. Physical randomisation is not really a thing, and KASLR not working without a PT_DYNAMIC ELF segment is intended.

closed 2022-04-12 by mintsuki