Limine-Bootloader/Limine #168: Limine returns wrong RSDP pointer in Virtualbox when booting with the Limine boot protocol

Glowman554, 2022-04-22:
**Limine version used**
Limine 3.4.1 bios

**Expected behaviour**
Limine returns the right RSDP pointer.

**Actual (broken) behaviour**
Limine returns incorrect RSDP pointer in Virtualbox when booting with Limine boot protocol, but the correct RSDP pointer is returned. when booting with stivale2. 

**To Reproduce**
Booting [this](https://github.com/limine-bootloader/limine/files/8539715/foxos.qcow2.tar.gz) image in virtualbox.
The code can be found [here](https://github.com/TheUltimateFoxOS/FoxOS) (the only modification is the print of the RSDP pointer)
Selecting the Limine protocol will return an incorrect RSDP pointer.
Selecting the Stivale2 protocol will return an correct RSDP pointer.

The RSDP pointer is printed in the serial log.

**Additional context**
Virtualbox Limine Protocol (Kernel serial log):
```
[main.cpp:66] Hello, world!
[main.cpp:67] Booting FoxOS using the limine boot protocol...
[main.cpp:68] rsdp: ffff8000000e0000
```

Virtualbox Stivale2 Protocol (Kernel serial log):
```
[main.cpp:66] Hello, world!
[main.cpp:67] Booting FoxOS using the stivale2 boot protocol...
[main.cpp:68] rsdp: e0000
```

mintsuki, 2022-04-22:
Hi,

This is not a bug. The pointers passed as part of the Limine boot protocol are all offset to the higher half direct map, as that makes them easier to handle. Stivale2 would do the same if the relevant bit in the header is set.

closed 2022-04-22 by mintsuki

Glowman554, 2022-04-22:
I have a question then. It seems that the rsdp pointer doesent get correctly mapped. Do you see something wrong with [this](https://github.com/TheUltimateFoxOS/horizon/blob/main/core/memory/memory.cpp#L60) mapping code? It works in qemu and even on real hardware (at least on the hardware i tested it on)

mintsuki, 2022-04-22:
Sorry, but this is not the place to ask me to debug your code. Ask in discord.gg/osdev if you're not there already, I am sure you'll find people to help you.