Limine-Bootloader/Limine #176: Limine fails when Font and Wallpaper are provided

kpect, 2022-05-28:
Limine fails to boot when `TERM_WALLPAPER` and `TERM_FONT` are provided -
[https://drive.google.com/file/d/1CkCVJkV9cBkNwCFYW_XxvriSM0P2CIuC/view?usp=sharing](https://drive.google.com/file/d/1CkCVJkV9cBkNwCFYW_XxvriSM0P2CIuC/view?usp=sharing)
If commented out limine boots fine - 
[https://drive.google.com/file/d/1HtTqo0_20NLAh0nf_sEOC7mggFd9I2ey/view?usp=sharing](https://drive.google.com/file/d/1HtTqo0_20NLAh0nf_sEOC7mggFd9I2ey/view?usp=sharing)

Here is the limine.cfg:
> ${KERNEL}=boot:///boot/vmlinuz
> ${ARGS}=root=/dev/sda1 ro vt.default_red=21,202,171,196,135,204,65,231,104,227,175,224,115,225,108,240 vt.default_grn=21,94,203,141,135,152,129,231,104,54,215,167,115,162,191,240 vt.default_blu=21,94,141,98,175,224,121,231,104,54,138,101,201,220,196,240
> 
> TERM_WALLPAPER=boot:///boot/splash.bmp
> TERM_FONT=boot:///boot/tosh-sat.f16
> TERM_FONT_SIZE=9x16
> INTERFACE_RESOLUTION=1600x900
> INTERFACE_BRANDING=***** AXE *****
> INTERFACE_BRANDING_COLOR=1
> DEFAULT_ENTRY=1
> GRAPHICS=yes
> TIMEOUT=3
> TERM_BACKDROP=262626
> TERM_FOREGROUND=A8A8A8
> TERM_BACKGROUND=51262626
> TERM_PALETTE=151515;CA5E5E;ABCB8D;C48D62;8787AF;CC98E0;418179;E7E7E7
> TERM_PALETTE_BRIGHT=686868;E33636;AFD78A;E0A765;7373C9;E1A2DC;6CBFC4;F0F0F0
> 
> :Boot
>         PROTOCOL=linux
>         KERNEL_PATH=${KERNEL}
>         KERNEL_CMDLINE=${ARGS} mds=full quiet
> 
> :Safe Mode
>         PROTOCOL=linux
>         KERNEL_PATH=${KERNEL}
>         KERNEL_CMDLINE=${ARGS} single

> $ doas fdisk -l
> Disk /dev/sda: 4 GiB, 4294967296 bytes, 8388608 sectors
> Disk model: VBOX HARDDISK
> Units: sectors of 1 * 512 = 512 bytes
> Sector size (logical/physical): 512 bytes / 512 bytes
> I/O size (minimum/optimal): 512 bytes / 512 bytes
> Disklabel type: gpt
> Disk identifier: C6ABA0D7-B303-794C-9064-E042D2983AD1
> 
> Device       Start     End Sectors  Size Type
> /dev/sda1     2048 7170047 7168000  3,4G Linux filesystem
> /dev/sda2  7170048 8386559 1216512  594M Linux swap
> $ uname -a
> Linux AXE 5.10.117 #2 SMP PREEMPT Wed, 25 May 2022 13:41:28 +0000 x86_64 GNU/Linux
> $ free -h
>                total        used        free      shared  buff/cache   available
> Mem:           993Mi        13Mi       949Mi       0,0Ki        29Mi       887Mi
> Swap:          593Mi          0B       593Mi

Link to the font:
[https://github.com/viler-int10h/vga-text-mode-fonts/blob/master/FONTS/PC-OTHER/TOSH-SAT.F16](https://github.com/viler-int10h/vga-text-mode-fonts/blob/master/FONTS/PC-OTHER/TOSH-SAT.F16)

Link to the wallpaper:
[https://drive.google.com/file/d/1WWNO1zgH9OF1OYf4iyfIAejMifU7foOy/view?usp=sharing](https://drive.google.com/file/d/1WWNO1zgH9OF1OYf4iyfIAejMifU7foOy/view?usp=sharing)

mintsuki, 2022-05-29:
Hi,

I cannot reproduce the issue. Using the provided font and background works fine for me.

Are you sure something else isn't the cause? Also, what Limine version is this?

kpect, 2022-05-29:
Limine version is 3.5
Looks like I've found the problem. It was related to `TERM_FONT` and `TERM_FONT_SIZE` settings. When I've changed it from **9x16** to **8x16**, everything began to work as expected.
The problem is that this error is not obvious. The ticket can be closed I believe.
Thanx.

mintsuki, 2022-06-19:
Closing this as unable to reproduce

closed 2022-06-19 by mintsuki