Limine-Bootloader/Limine #184: Testers reporting invalid opcode

bkauler, 2022-06-25:
I have built EasyOS 4.2 usb-stick image with Limine. Previously, had used rEFInd for UEFI booting and Syslinux for legacy-BIOS booting. The promise of Limine, to integrate both, was too good to resist.

Limine works fine on four computers I tested, oldest is a Compaq Presario; however, after releasing Easy 4.2, I am receiving feedback of Limine panics.

Here is my blog announcement for 4.2:

https://bkhome.org/news/202206/easyos-dunfell-series-version-42-released.html

Here is one of the feedbacks, email from Alfons. Looks like he has manually typed in the panic report from the screen:

-----------------------------------------
PANIC: Invalid opcode
Stacktrace
[OxeBae]
‹panic+Ox9d>
[Oxe310].
Kexcept+0x1d>
[0xe57e] (raise exception 6+0xe›
[Oxeceb]
<: udivmoddi4+0xb>
[@xd614] <gpt get
guid+0x2b>
[0x8af4] ‹disk create index+0x3bb>
[Bxe1f5] (entry+Bx3f>
End of trace. Press
CTRL+ALT+DEL
to
reboot

MAC (works fine)
 Startup successful with:
 Processor              : 4x Intel(R) Core(TM) i7-2620M CPU @ 2.70GHz

DELL and ASUS (PANIC: Invalid opcode)
 Startup not successful with:
 Processor              : 2x Intel(R) Core(TM)2 Duo CPU     T7500  @ 2.20GHz
 Processor              : 2x Intel(R) Core(TM)2 Duo CPU     E7600  @ 3.06GHz
--------------------------------

I have more of these, from other testers. So I guess the question is, need to modify the assembly language to not use the invalid op-code.

A very long time ago, like around 1995, I used to do x86 assembly programming. I could take a look at the code, but then you guys would be more familiar with it. So would appreciate if you could fix this.

mintsuki, 2022-06-26:
Hello, I do believe we fixed this issue in version 3.7.1 onwards. Is it possible they may try that one instead?

bkauler, 2022-06-29:
Alfons has sent email feedback, Limine no longer giving invalid opcode panic.
Good, can close this.

closed 2022-06-29 by bkauler