Limine-Bootloader/Limine #185: Tester reports a non usable e820 entry

bkauler, 2022-06-25:
Here is another panic report for EasyOS 4.2, using Limine 3.7. Email from Reynaldo:

```
Have encountered this problem on Easy 4.2 on my Fujitsu Futro MAS74 laptop during boot:

PANIC: A non usable e820 entry is inside a usable section.
[0xc5134686] <panic+0x86>
[0xc512966d] <sanitise_entries+0x38d>
[0xc51299cb] <memmap_alloc_range.part.0.constprop.0+0x10b>
[0xc512a40b] <ext_mem_alloc_type_aligned+0xab>
[0xc51212f1] <get_edid_info+0x21>
[0xc51225a9] <init_gop+0x479>
[0xc5136a21] <gterm_init+0xb1>
[0xc512c881] <term_vbe+0x51>
[0xc51346b4] <panic+0xb4>
[0xc512966d] <sanitise_entries+0x38d>
[0xc5129b3b] <memmap_alloc_range.part.0.constprop.1+0x11b>
[0xc5126130] <ext2_parse_dirent+0x3c0>
[0xc5126200] <ext2_open+0xa0>
[0xc5126627] <fopen+0x197>
[0xc513432e] <url_open0xae>
[0xc51377ac] <linux_load+0x3c>
[0xc5128738] <boot+0x1c8>
[0xc5129190] <_menu+0x9d0>
End of trace. System halted.
```

That error message means nothing to me. Are you able to help?

See also my other bug report, with link to Easy 4.2 blog announcement:
https://github.com/limine-bootloader/limine/issues/184

mintsuki, 2022-06-26:
Hello. 0836a62 *may* address the issue, not sure though because I can't reproduce and I have never seen it happen before.

In any case, I will go ahead and release 3.8.2, integrating this commit.

This release should also fix #184 as that was an issue specific to 3.7 that was fixed on 3.7.1 onwards.

bkauler, 2022-06-29:
Reynaldo has tested Limine with the above fix. His email to me:

> EasyOS 4.2.2 hangs during boot on my Fujitsu laptop. I`m seeing only the "Black Screen of Death."

...I presume he means not seeing any panic message either.

bkauler, 2022-07-05:
Here is an report from Reynaldo. He has tested EasyOS 4.2.3, which has Limine 3.10.1. Extract from his email:

> This what I am getting on USB stick fresh install of version 4.2.3 on 1st boot for Fujitsu:
> 
> "Limine 3.10.1
> Config file not found.
> For information on the format of Limine config entries, consult CONFIG.md in
> the root of the Limine source repository.
> Press a key to enter the Limine console..."

limine.cfg and limine.sys are in the first partition top-level on the usb-stick, it is a 7MiB fat12 with esp and boot flags set. And of course has /EFI/BOOT/BOOTX64.EFI

I'm wondering, Reynaldo said it is an old computer. A wild thought, but maybe it needs a bit more time to recognize the usb partition.
Though, that shouldn't be an issue. We have tested on some pretty old PCs, and OK. Oldest I tested was a Compaq Presario.

mintsuki, 2022-07-05:
Given the inconsistencies of these reports, couldn't it also be possible that the USB stick is damaged and that leads to issues with reading from it?

bkauler, 2022-07-06:
Reynaldo has replied. He has another computer that works with the same usb-stick.
He has now tested with three usb-sticks:

> Defective USB stick is not an issue because I have tried 3 diff. sticks, same problem occurred  for 4.2.3.

> This is the nearest Specs for my Fujitsu Lifebook series:

> [https://icecat.biz/rest/product-pdf?productId=37274458&lang=en](https://icecat.biz/rest/product-pdf?productId=37274458&lang=en)

mintsuki, 2022-07-06:
Not sure. Without more extensive testing or reporting it's hard to tell what is going on with this particular issue.

> I did a search for "Fujitsu Futro MAS74", can't find it.

The S is supposed to be a 5. I have found this machine online.

Hopefully this issue is limited to this machine and no one else stumbles upon it. If that is not the case please let me know.

If this issue is unique to this machine I could still consider getting a used MA574 if I can find one to see if I can reproduce the problem in person.

bkauler, 2022-07-06:
Have received an email from Reynaldo:

> Here`s another info:
> 1.  When I enter to Limine Console command "editor" and Press F10 for reboot this is what I get:

[image]

> Above pix is the result when I press F10 for reboot in Limini command EDITOR.

> and pix below is the result for List Volumes command:

[image]

> and it hangs after that.

This is the entry in the usb-stick:

```
:EasyOS dunfell
    COMMENT=EasyOS Dunfell bootup
    PROTOCOL=linux
    TEXTMODE=yes
    KERNEL_PATH=boot://2/easyos/vmlinuz
    MODULE_PATH=boot://2/easyos/initrd
    KERNEL_CMDLINE=initrd=initrd rw wkg_uuid=41e4e492-f6a0-11ec-9a02-287fcfeb4376 wkg_dir=easyos/
```

bkauler, 2022-07-06:
Putting in a question of my own here:
Someone on the Puppy Forum informed me that "initrd=initrd" is not required on the kernel commandline, as it is already specified by MODULE_PATH.
Is that correct? I find out of course.

mintsuki, 2022-07-06:
> Putting in a question of my own here: Someone on the Puppy Forum informed me that "initrd=initrd" is not required on the kernel commandline, as it is already specified by MODULE_PATH. Is that correct? I can find out of course.

I don't know. What I can tell you is that `initrd=initrd` in this case is passed as-is to the kernel. Limine does not process the kernel command line. The kernel usually would pass stuff it itself doesn't recognise onto the initramfs.

`MODULE_PATH` loads 1 or more (if assigned multiple times) modules into memory, so when the kernel is started, whichever initramfs is specified as `MODULE_PATH` is already in memory.

bkauler, 2022-07-28:
Reynaldo has sent me some snapshots, still not working on his Fujitsu laptop. Trying to boot EasyOS 4.2.7, I presume from usb-stick, with Limine 3.12.1.

[image]

[image]

I have asked him to report directly to here in future, instead of going via me.

mintsuki, 2022-07-28:
It would appear that the MBR (or GPT?) on the USB stick is not getting recognised (again).

What I can think of is to remove checking for the boot signature as part of the MBR detection routine and seeing what happens...

Reynaldo333, 2022-08-02:
For the new EasyOS 4.3 version, I encountered `the black screen of death` same as in version 4.2.2.  It hangs and not seeing any panic messages.

mintsuki, 2022-08-02:
Hi.
EasyOS 4.3 does not actually ship with the latest Limine version. The version reported by `limine-version` is 3.13, and the actually installed (deployed as bootloader) version is 3.12.1. The latest version is 3.14.

Please either you or Barry ensure that you are on the latest version and see if you still encounter this.

bkauler, 2022-08-03:
Sorry about that, only having 3.12.1.

There has been a very interesting development. Reynaldo has sent me an email that EasyOS 32-bit version 4.3.1, compiled for a i686 CPU, WORKS on the Fujitsu.
He sent a photo:
[image]

Now for the interesting news: it only has Limine 3.12.1

All of the packages for the 32-bit EasyOS were compiled in OpenEmbedded, except for Limine, that I compiled in a running 32-bit EasyOS, package here:

https://distro.ibiblio.org/easyos/x86/packages/pet/pet_packages-dunfell/

So, there are two possibilities. Maybe the OpenEmbedded cross-compile environment has somehow broken limine.sys or limine-deploy

Or, compiling them in a 32-bit environment has somehow fixed limine.sys or limine-deploy.

One thing I noticed: the cross-compiled files are slightly bigger. Example, limine.sys is 172K, versus 152K compiled in running 32-bit EasyOS.

But, that doesn't explain why the cross-compiled Limine has worked on every other computer, the Fujitsu is currently the only holdout.

bkauler, 2022-08-03:
The 32-bit EasyOS is announced here:

https://bkhome.org/news/202208/easyos-32-bit-dunfell-series-version-431.html

bkauler, 2022-08-03:
Correction: I just checked how I created the 32-bit Limine package.
I used your binary package, so didn't create limine.sys, only compiled limine-deploy. So that 152K size is your file.
This was in a running 32-bit easyOS.

mintsuki, 2022-08-03:
What I think is more likely happening here is that, since 32-bit easyOS ships without a UEFI executable, it boots via BIOS, and that works fine instead.

bkauler, 2022-08-03:
Oh, I thought that Fujitsu Futro MAS74 is an old BIOS-firmware computer.
So maybe it is UEFI, will check with Reynaldo.

bkauler, 2022-08-04:
Reynaldo sent me some info about the firmware, it is "Phoenix SecureCore", dated 7/6/2016 and is UEFI.
The laptop has Intel i5 CPU and 4GB RAM.
I think it is 32-bit UEFI.

mintsuki, 2022-08-04:
> Reynaldo sent me some info about the firmware, it is "Phoenix SecureCore", dated 7/6/2016 and is UEFI. The laptop has Intel i5 CPU and 4GB RAM. I think it is 32-bit UEFI.

Knowing that it was 32-bit UEFI has been invaluable to detecting and fixing an issue with regards to 32-bit UEFI that I never caught due to the rarity of 32-bit UEFI (and therefore me neglecting it more than the other ports).

I do not know if this fix will make Limine work on Reynaldo's machine since I am not even sure it is actually a 32-bit UEFI machine, but it has been a good fix to have nonetheless.

Upstreamed in 3.14.1.

bkauler, 2022-08-05:
That's great!

I read some specs on the Fujitsu Futro MA574 that it has 32-bit Windows.
Apparently, 32-bit Windows must have 32-bit UEFI.

However,  a 64-bit Linux kernel with CONFIG_EFI_MIXED enabled, can boot on a 32-bit UEFI.

Reynaldo333, 2022-08-07:
Thank you guys,
The new EASYOs 4.3.2 is working now on my Fujitsu laptop.

Best regards,
Reynaldo

bkauler, 2022-08-08:
Reynaldo, 
I have replied to you by email. 4.3.2 a 32-bit release, that only supports legacy-BIOS booting. We have already found limine is ok with that.

64-bit EasyOS 4.3.2 is coming soon, likely in a day or two. It will have limine 3.14.2.

mintsuki, 2022-08-08:
By the way, it goes without saying that for 32-bit UEFI booting, the `BOOTIA32.EFI` file must be present, just like `BOOTX64.EFI` is for 64-bit UEFI.

Reynaldo333, 2022-08-09:
Hello guys,

The latest 64-bit EasyOS 4.3.3 still did not work on my Fujitsu.  It hangs and no panic messages.

bkauler, 2022-08-09:
Reynaldo,
In your email to me, you have informed that you used Rufus.
Do not use Rufus. Write the image file to the drive using 'dd' or 'easydd' in Linux, or one of the Windows tools as listed here:

https://easyos.org/install/how-to-install-easyos-on-a-new-ssd.html

In Windows, I have experience with USB Image Tool:

https://easyos.org/install/how-to-write-easyos-to-a-flash-drive.html#USB_Image_Tool

Reynaldo333, 2022-08-10:
Barry,
Done writing the image file using _USB Image Tool 1.8.5_ and _Balena Etcher 1.7.9_--same results. Fujitsu hangs and no panic message displayed.

@mintsuki, 
Is there another OS using Limine aside EasyOS for me to try on my Fujitsu?

mintsuki, 2022-08-10:
You can try this OS: https://github.com/lyre-os/lyre/releases/download/nightly-2022-08-10/lyre-nightly-20220810.iso

Additionally, it would probably be wise to remove the `QUIET=yes` flag and add `VERBOSE=yes`, just so we get an actual output from the bootloader rather than a blank screen. That can be changed in `limine.cfg` for EasyOS.

Reynaldo333, 2022-08-10:
I edited the **limine.cfg** for EasyOS as what you`ve told.  Still it hangs and no panic messages displayed.

**For lyre-os** it also hangs after these messages:
**PANIC**: A non-usable memory map entry is inside a usable section.
Stacktrace:
  [0xc5127f54] <panic+0x94>
  [0xc5121c81] <sanitise_entries+0x3b1>
  [0xc5122bf5] <pmm_reclaim_uefi_mem+0x305>
  [0xc5122cc0] <get_memmap+0x20>
  [0xc5138d29] <limine_load+0xd290>
  [0xc513d756] <boot+0x106>
  [0xc513f1a2] <_menu+0x19b2>
End of trace. System halted.

mintsuki, 2022-08-12:
Reynaldo, can you please try the following ISO and report back?
https://cdn.discordapp.com/attachments/706151776495206426/1007608579852865566/test.iso

Reynaldo333, 2022-08-13:
mintsuki,
Done writing the test.iso using **USB** Image Tool and **EasyDD**. Both displayed same results.
It loads the Limine editor with a very nice desktop background then run some functions and hangs after the ff.

> Terminal feature, revision 0
> 1 terminal (s)
> columns: 137
> rows: 40
> Using frame buffer: 0xffff830fccf69000
> Write function at: 0xc512c6f0

mintsuki, 2022-08-13:
Then it would be interesting to now try easyOS with 3.15.1, and see what happens.

You can mount the easyOS image and replace the `BOOTX64.EFI` and `BOOTIA32.EFI` files in the ESP with the ones from Limine 3.15.1.

Reynaldo333, 2022-08-13:
mintsuki, It works! Thank you very much.

 DSC_0022.JPG
<https://drive.google.com/file/d/1QSKA9rOsWGM-2ZDpeaBsM5JFsRdDgz4Y/view?usp=drive_web>

Best regards,
Reynaldo

Reynaldo333, 2022-08-13:
It works! Thank you.

Best regards,
Reynaldo

mintsuki, 2022-08-14:
That's great! I hope we can close this issue for good now.

closed 2022-08-14 by mintsuki

Reynaldo333, 2022-08-22:
Hi,
Has an issue on USB port 2.0. It hangs and displays *config file not found*
and it won't boot back on USB 3.0 port. USB stick is unmountable after that.
It has been a week that I am using USB 3.0 on the latest 64-bit Easy OS
without any issues.

For the 32-bit Easy OS, no issues on USB 2.0 and USB 3.0 ports.

mintsuki, 2022-08-22:
Please open another issue for this.

commit 0836a62f (2022-06-26): pmm: Minor overhaul to try and fix #185
commit d98b1e98 (2022-06-26): pmm: Minor overhaul to try and fix #185