Limine-Bootloader/Limine #187: Unable to compile 3.8.2

bkauler, 2022-06-26:
I compiled 3.7 and earlier OK. However, with 3.8.2 get:

`configure: error: cannot find sources (Limine) in ../limine-3.8.2 or ..`

...that's when configuring in the OpenEmbedded/Yocto environment.

I am cross-compiling in OpenEmbedded/Yocto, which does the build in a separate directory outside of the source. You might be interested to see the recipe:

```
# Recipe created by recipetool
# recipetool create -o limine_3.8.2.bb https://github.com/limine-bootloader/limine/releases/download/v3.8.2/limine-3.8.2.tar.gz

SUMMARY = "x86/x86_64 BIOS/UEFI bootloader"
HOMEPAGE = "https://limine-bootloader.org/"

LICENSE = "UCB & GPLv3"
LIC_FILES_CHKSUM = "file://LICENSE.md;md5=566da3e4be60cd868e42cb88c9c0ddd3 \
                    file://stivale/LICENSE.md;md5=21e2e2557201f26b907e5694c9e179b9 \
                    file://libgcc-binaries/COPYING.RUNTIME;md5=fe60d87048567d4fe8c8a0ed2448bcc8 \
                    file://libgcc-binaries/COPYING;md5=64305ff51f34d0aca60d187e21a09d59 \
                    file://libgcc-binaries/COPYING3;md5=d32239bcb673463ab874e80d47fae504 \
                    file://limine-efi/COPYING;md5=c27a4b4a954b36c8afddf7587fd749be \
                    file://freestanding_headers/LICENSE.md;md5=9621911223dde43748ad798e3804a3ed"

SRC_URI = "https://github.com/limine-bootloader/limine/releases/download/v${PV}/limine-${PV}.tar.gz \
           file://remove-ext4-encrypt-warning.patch"

SRC_URI[md5sum] = "b1076ccefa6412d52b422ef4da5881b8"
SRC_URI[sha256sum] = "08b2675e3190d2670668d42fac7a239dd7e1a3ccd44da44506623a50dc373ab1"

inherit autotools

# need 'paste' in coreutils...
# dont think need gnu-efi as it is builtin to limine source... ditto zlib
DEPENDS = "gnu-efi mtools mtools-native nasm-native zlib-native zlib coreutils-native"

EXTRA_OECONF = ""
```

...showing that as I have an extra question. Those dependencies could be reduced. The ones with "-native" run in the host OS and are required for the build. So I guess nasm-native is no longer required?

I changed a line in the recipe:

`inherit autotools-brokensep`

...which causes the build to happen in the source code, but still no go:

`configure: error: cannot find sources (Limine) in . or ..`

Do you know what has changed since 3.7 that could be causing this configure failure?

Note, target is gcc 9.3.0, configured for nocona CPU (one of the earlier x86_64) and glibc 2.33.

bkauler, 2022-06-26:
OK, I testing compiling earlier versions in OpenEmbedded:

3.7.1 OK
3.7.3 OK
3.8 FAIL

...failure is that same configure error.

I'm looking at the source and I can't see why configure is failing. What has changed between 3.7.3 and 3.8?

mintsuki, 2022-06-26:
Hello. In preparation for the aarch64 port, the build system has been slightly changed starting in 3.8.

By default, it does not build *anything* as described in the README.md. You'll probably want to pass  to configure in order to build all ports (IA-32-UEFI, x86-64-UEFI, and BIOS) as before.

mintsuki, 2022-06-26:
gnu-efi is not needed as Limine does not use gnu-efi, but rather it's built-in limine-efi fork. A `nasm` that runs on the machine performing the build is still required. No runtime dependency is necessary (other than `glibc` for `limine-deploy`)/

mintsuki, 2022-06-26:
So I guess this line:
```
DEPENDS = "gnu-efi mtools mtools-native nasm-native zlib-native zlib coreutils-native"
```
could be shortened to
```
DEPENDS = "mtools-native nasm-native coreutils-native"
```

mintsuki, 2022-06-26:
Out of tree builds are not broken.

bkauler, 2022-06-29:
I guess we can close this.

bkauler, 2022-07-02:
I stayed on 3.7.3, until now, as want that new 3.9.2! Still have the same problem with configure saying it cannot find the source. I examined the log, and see also it complained that version.sh is missing.

I messed around a bit, found a version.sh online and tried to use that, no go. OpenEmbedded rebuilds the configure script. I changed the recipe so that it will use the existing configure script, and that works. In a cross-compile environment, it is probably best to let OE do its default behaviour; however, using the existing configure does work.

Here is the working recipe:

```
# Recipe created by recipetool
# recipetool create -o limine_3.9.2.bb https://github.com/limine-bootloader/limine/releases/download/v3.9.2/limine-3.9.2.tar.gz

SUMMARY = "x86/x86_64 BIOS/UEFI bootloader"
HOMEPAGE = "https://limine-bootloader.org/"

LICENSE = "UCB & GPLv3"
LIC_FILES_CHKSUM = "file://LICENSE.md;md5=566da3e4be60cd868e42cb88c9c0ddd3 \
                    file://stivale/LICENSE.md;md5=21e2e2557201f26b907e5694c9e179b9 \
                    file://libgcc-binaries/COPYING.RUNTIME;md5=fe60d87048567d4fe8c8a0ed2448bcc8 \
                    file://libgcc-binaries/COPYING;md5=64305ff51f34d0aca60d187e21a09d59 \
                    file://libgcc-binaries/COPYING3;md5=d32239bcb673463ab874e80d47fae504 \
                    file://freestanding-headers/LICENSE.md;md5=9621911223dde43748ad798e3804a3ed \
                    file://limine-efi/COPYING;md5=c27a4b4a954b36c8afddf7587fd749be"

SRC_URI = "https://github.com/limine-bootloader/limine/releases/download/v${PV}/limine-${PV}.tar.gz \
           file://remove-ext4-encrypt-warning.patch"

SRC_URI[md5sum] = "b1f6f2e46d5d1ec4476ae3ff9df33079"
SRC_URI[sha256sum] = "546cff6c94e40c7996581b60e350b286321463bc282dabf111e0ed6d796fa6e9"

inherit autotools

DEPENDS = "mtools-native nasm-native coreutils-native"

EXTRA_OECONF = "--enable-all"

#complains that version.sh missing, try hack...
XXXdo_configure_prepend() {
 cp -f ${WORKDIR}/version.sh ${S}/
 sed -i '/version\.sh/d' ${S}/GNUmakefile.in
}

#give up rebuilding with autotools, just run existing configure script...
do_configure() {
 oe_runconf
}

```

closed 2022-07-02 by mintsuki

mintsuki, 2022-07-02:
If all is well, this along with the addition of support for `fslabel://` should be up in 3.10

mintsuki, 2022-07-02:
There was a minor issue with the `version.sh` script in 3.10. Fixed in 3.10.1

commit 2f0d48ab (2022-07-02): build: Undo changes that made reconf impossible in tarball releases. Fixes #187
commit 8f8af188 (2022-07-02): build: Make sure autoreconf -fvi is able to reconf tarballs. Fixes #187
commit ddafc630 (2022-07-02): build: Undo changes that made reconf impossible in tarball releases. Fixes #187
commit fef3c2c4 (2022-07-02): build: Make sure autoreconf -fvi is able to reconf tarballs. Fixes #187