Limine-Bootloader/Limine #188: Unable to chainload Windows 10

bkauler, 2022-07-01:
I'm running EasyOS (Linux) on my Mele mini-PC, installed on a SATA SSD, with Limine bootloader 3.7.3. 

The Mele came with Windows 10 in eMMC memory, so the partitions are /dev/mmcblk0p1, /dev/mmcblk0p2 etc. 

I am trying to add an entry in limine.cfg to chainload Windows. Earlier test with rEFInd works:

```
menuentry "Windows 10" {
 volume SYSTEM
 loader \EFI\Microsoft\Boot\bootmgfw.efi
}
```
Finding the filesystem uuid:

```
# blkid /dev/mmcblk0p1
/dev/mmcblk0p1: LABEL="SYSTEM" UUID="5A85-A404" TYPE="vfat"
```
Creating entry in limine.cfg:

```
:Windows 10 in mmcblk0
    COMMENT=Windows 10 in mmcblk0
    PROTOCOL=chainload
    #IMAGE_PATH=guid://5A85-A404/EFI/Microsoft/Boot/bootmgfw.efi
    IMAGE_PATH=hdd://1:1/EFI/Microsoft/Boot/bootmgfw.efi
```
You can see the two attempts, and also tried back-slashes.
But always Limine panics and reports cannot find that .efi file.

I suppose the first question is, does Limine recognise /dev/mmcblk<number> devices?

mintsuki, 2022-07-01:
Hello. Please, can you try using the GPT partition GUID (partuuid) value for the FAT partition? Limine supports partition GUIDs and actual filesystem UUIDs (like for ext4), but not FAT filesystem IDs.

It's possible to obtain the partuuid of a GPT partition by listing the files in `/dev/disk/by-partuuid` on Linux, usually.

mintsuki, 2022-07-01:
> I suppose the first question is, does Limine recognise `/dev/mmcblk<number>` devices?

That's really wholly up to the BIOS/firmware, but I don't see why it wouldn't recognise them, provided the firmware supports them.

mintsuki, 2022-07-01:
Also I'd like to take the chance to mention that I am really thankful that you decided to use Limine as the EasyOS bootloader, and that I've been surfing the Puppy Forums to see if everything was going smoothly.

I noticed the post with [this](
https://forum.puppylinux.com/download/file.php?id=10001) image attached. This is a bug that was not new to me, so I decided to investigate further with the help of a friend and I'm happy to say that we were able to track down the issue and that it is fixed in 3.9.2!

bkauler, 2022-07-02:
What about PARTUUID in the format `<NT disk identifier>-<partition number>` for mbr drives?

The Linux kernel recognizes this format. For example:
```
# fdisk -l /dev/sda
 ...
Disk identifier: 0xfe17e1bd
```
Then for the kernel commandline, specifying partition 2, which must be two digit hexadecimal:

`root=PARTUUID=fe17e1bd-02`

mintsuki, 2022-07-02:
How about adding a `label://` root with a unified labels namespace?

closed 2022-07-02 by mintsuki

bkauler, 2022-07-02:
fslabel, that's great. I can also confirm that both of these work, guid and hdd:

```
:Windows 10 in mmcblk0
    COMMENT=Windows 10 in mmcblk0
    PROTOCOL=chainload
    IMAGE_PATH=hdd://2:1/EFI/Microsoft/Boot/bootmgfw.efi
    #IMAGE_PATH=guid://09ddd6ad-80b0-4265-8d7f-1621eb5a4d9f/EFI/Microsoft/Boot/bootmgfw.efi
```

hdd didn't work for me before because I thought the eMMC would be the number 1 drive.