Limine-Bootloader/Limine #191: EFI chainload enhancement request

bkauler, 2022-07-08:
Currently, the format is this:

```
:Internal drive
   PROTOCOL=chainload
   guid://${PARTUUID}/path-to-efi
```

It would be good to support the last parameter left off:

```
:Internal drive
   PROTOCOL=chainload
   guid://${PARTUUID}/
```

Like you support for mbr chainloading, can leave off the PARTITION parameter and it just chainloads the mbr.

Same can be done for efi chainloading, just let the uefi do it's default finding of a efi file.

bkauler, 2022-07-08:
By "drive-uuid" I mean a uuid that identifies the drive, not a partition, like this:

```
# gdisk -l /dev/nvme1n1
 ...
Disk identifier (GUID): 494D4539-FF62-4BB1-B06B-86CC0BCB8B14
```
And also get it from lsblk:

# lsblk --pairs -o PTUUID --nodeps /dev/nvme1n1
PTUUID="494d4539-ff62-4bb1-b06b-86cc0bcb8b14"

And I think that is more correct, with lower-case letters.

mintsuki, 2022-07-08:
Well having such a chainloading option is really outside of the scope of how Limine is meant to operate.

That said, it would be possible, and it is something I meant to implement for a while, for both UEFI and BIOS to have a "return to firmware"/"boot next drive" option, which would work in a similar way as described, without having to special case a chainload path or anything like that.

bkauler, 2022-07-09:
Another related question. It is ok to chainload modern Windows, like this:

```
:Windows (drive nvme1n1)
    COMMENT=Windows bootup
    PROTOCOL=chainload
    IMAGE_PATH=guid://67c45f63-8668-4fbd-9aa9-341a6dae89b4/EFI/Microsoft/Boot/bootmgfw.efi

```
However, Windows XP requires running an executable 'ntldr', and I cannot see how Limine can do that. Running "file ntldr" shows it to be "DOS executable (COM)".

Also, I think that Windows 7 requires running a different executable, 'bootmgr'

There are lots of people still using Win7, so probably something worthwhile supporting.

mintsuki, 2022-07-09:
Windows XP does not support UEFI as far as I am aware. NTLDR is a BIOS loader so it'd only work with BIOS chainloading.

As for Windows 7, I know it supports UEFI in a limited way, but that kind of goes outside the scope of this issue.

commit 95272c5f (2022-07-10): protos: Add chainload_next protocol. Closes #191
commit d351ed87 (2022-07-10): protos: Add chainload_next protocol. Closes #191
closed 2022-07-10 by mintsuki