Limine-Bootloader/Limine #192: Limine should show name and version on boot selection screen

zrudyt, 2022-07-09:
You have built a great utility, so why not proudly advertize it?  I think Limine and the version number should show somewhere on oeither the top or bottom blank area on the screen,

mintsuki, 2022-07-09:
Hm, well, it does?

mintsuki, 2022-07-09:
Oh and by the way thanks for the compliments, appreciated!

zrudyt, 2022-07-09:
Sorry, my post wasn't clear because I forgot a word ;-)  "I  think Limine and the version number should **ALWAYS** show ...".  If I set INTERFACE_BRANDING, the name and version disappears.

The compliment is well deserved.  I've been using limine since Barry K mentioned it on hos Puppy blog. I even wrote a quick and dirty script to complie limine-deploy on linux and use it to update my USB stick, then copy limine.sys.  No typing or manual intervention.

mintsuki, 2022-07-09:
Ah, I understand now. Well, I like to give people the option to customise what shows in the menu's title, even if it means not showing "Limine <version>".

As for crediting, in the console (which should be always accessible whenever the menu is) you can show the version and copyrighting information.

mintsuki, 2022-07-09:
Yeah I get what you mean but there would undoubtedly be people who wouldn't be happy with that decision; plus people can still fork/patch the branding away in source code themselves.

I am personally fine with attribution and copyright being accessible from the unintrusive console, so it's not a big deal for me.

Thanks for all the kind words once again, though.

closed 2022-07-09 by mintsuki