Limine-Bootloader/Limine #193: Unable to cross-compile 3.11

bkauler, 2022-07-11:
As usual, cross-compiling in OpenEmbedded. The last version that I compiled was 3.10.1, successfully. With 3.11:

```
| checking for i686-elf-gcc... no
| checking for clang... no
| checking for gcc... yes
| checking for i686-elf-addr2line... no
| checking for llvm-addr2line... no
| checking for addr2line... no
| configure: error: no suitable CROSS_ADDR2LINE
```

bkauler, 2022-07-11:
I have narrowed it down. Version 3.10.2 fails, same error as above.
What has happened between 3.10.1 and 3.10.2 that is causing this?

bkauler, 2022-07-11:
OK. But, there is no quick fix. Tried adding deps:

`DEPENDS = "mtools-native nasm-native coreutils-native binutils-native binutils-cross-x86_64"`

It shouldn't be required to add those as deps, as they are already there. OE is a cross-compile environment, and everything is there.
It would seem that the external cross-detect is mis-detecting.

bkauler, 2022-07-11:
Interesting. OE sets up a build environment for each package, and creates:

`build-amd64/tmp/work/nocona-64-poky-linux/limine/3.11-r9/recipe-sysroot-native/usr/bin/x86_64-poky-linux`

With these executables:

```
x86_64-poky-linux-addr2line
x86_64-poky-linux-ar
x86_64-poky-linux-as
x86_64-poky-linux-c++
x86_64-poky-linux-c++filt
x86_64-poky-linux-cpp
x86_64-poky-linux-dwp
x86_64-poky-linux-elfedit
x86_64-poky-linux-g++
x86_64-poky-linux-gcc
x86_64-poky-linux-gcc-9.3.0
x86_64-poky-linux-gcc-ar
x86_64-poky-linux-gcc-nm
x86_64-poky-linux-gcc-ranlib
x86_64-poky-linux-gcov
x86_64-poky-linux-gcov-dump
x86_64-poky-linux-gcov-tool
x86_64-poky-linux-gprof
x86_64-poky-linux-ld
x86_64-poky-linux-ld.bfd
x86_64-poky-linux-ld.gold
x86_64-poky-linux-nm
x86_64-poky-linux-objcopy
x86_64-poky-linux-objdump
x86_64-poky-linux-ranlib
x86_64-poky-linux-readelf
x86_64-poky-linux-size
x86_64-poky-linux-strings
x86_64-poky-linux-strip
```
Looking around I found some symlinks to the above, ar, as, cpp, gcc, etc., here:

`build-amd64/tmp/work/nocona-64-poky-linux/limine/3.11-r9/recipe-sysroot-native/usr/libexec/x86_64-poky-linux/gcc/x86_64-poky-linux/9.3.0`

But cannot find addr2line symlinked from anywhere.

I could try a hack, create symlink, prior to the reconf, but it really seems like the external cross-check is over-checking in the situation being inside OE.

mintsuki, 2022-07-11:
Or you could just pass `TOOLCHAIN=x86_64-poky-linux` to Limine's configure :)

bkauler, 2022-07-11:
Good, setting CROSS_TOOLCHAIN worked. Here is the OpenEmbedded recipe, in case anyone is interested:

```
# Recipe created by recipetool
# recipetool create -o limine_3.11.bb https://github.com/limine-bootloader/limine/releases/download/v3.11/limine-3.11.tar.gz

SUMMARY = "x86/x86_64 BIOS/UEFI bootloader"
HOMEPAGE = "https://limine-bootloader.org/"

LICENSE = "UCB & GPLv3"
LIC_FILES_CHKSUM = "file://LICENSE.md;md5=566da3e4be60cd868e42cb88c9c0ddd3 \
                    file://stivale/LICENSE.md;md5=21e2e2557201f26b907e5694c9e179b9 \
                    file://libgcc-binaries/COPYING.RUNTIME;md5=fe60d87048567d4fe8c8a0ed2448bcc8 \
                    file://libgcc-binaries/COPYING;md5=64305ff51f34d0aca60d187e21a09d59 \
                    file://libgcc-binaries/COPYING3;md5=d32239bcb673463ab874e80d47fae504 \
                    file://freestanding-headers/LICENSE.md;md5=9621911223dde43748ad798e3804a3ed \
                    file://limine-efi/COPYING;md5=c27a4b4a954b36c8afddf7587fd749be \
                    file://cross-detect/LICENSE.md;md5=9621911223dde43748ad798e3804a3ed"

SRC_URI = "https://github.com/limine-bootloader/limine/releases/download/v${PV}/limine-${PV}.tar.gz \
           file://remove-ext4-encrypt-warning.patch"

SRC_URI[md5sum] = "536aeb5f52f326f7cc83c50772e1cab1"
SRC_URI[sha256sum] = "757b5765597fde3cf08fd6f2a1e22e2bc1792458972baf5305fa4fe1782d3309"

inherit autotools

DEPENDS = "mtools-native nasm-native coreutils-native"

export CROSS_TOOLCHAIN="x86_64-poky-linux"

EXTRA_OECONF = "--enable-all"
```

closed 2022-07-11 by bkauler

bkauler, 2022-07-11:
Just for the record, OE already has this defined:

`TARGET_SYS = "x86_64-poky-linux"`

So it would be better to assign $TARGET_SYS to CROSS_TOOLCHAIN in the recipe.