Limine-Bootloader/Limine #194: Panic on v3.11 - boots OK on v3.10.5 (Dell Latitude E7250)

zrudyt, 2022-07-11:
I see a PANIC: Double Fault on boot with v3.11 - stack trace is attached.  v3.10.5 booted fine on this computer.  I always install the same way from the binary package, compile and run limine-deploy on linux, then copy limine.sys.  I boot from USB stick /dev/sdb1
[image]

mintsuki, 2022-07-11:
If I were to guess I'd say it's related to `77616a0`. Though I'm not sure how it would cause this issue, the commit seems pretty harmless.

Given it's not an essential change, I will revert it on 3.0-branch, for now.

zrudyt, 2022-07-11:
Let me know if you want me to do some testing or need any further info.

zrudyt, 2022-07-11:
3.11.1 boots fine - thanks for the quick turnaround.  My offer still stands if you need me to do some testing for you to help narrow down the root cause

mintsuki, 2022-07-12:
Thanks for the help offer! I will let you know by mentioning you if I need some help.

Closing this for now as this is no longer an issue.

closed 2022-07-12 by mintsuki