Limine-Bootloader/Limine #195: fslabel panic on Dell D630 and Asus PC

bkauler, 2022-07-13:
Alfons has been testing Limine on different computers and emailing reports to me.
Today I received this email, testing 3.11.1:

> hmm, the same problem as with 3.10.1
> Details see attachment.
> affected by this:
> - Dell D630
> - PC ASUS

[image]
[image]

I have asked him to confirm that fslabel does work on other computers with 3.11.1, and also whether it worked on the Dell and Asus at an earlier version of Limine.

Alfons has been doing a lot of testing, so knows what he is doing and the path to vmlinuz would be correct.

bkauler, 2022-07-13:
I just remembered, this is a continuation of #189 
And there he reported fslabel does work on his Mac, not the Dell.

mintsuki, 2022-07-13:
It is really really really odd that `fslabel` would work on one machine and fail or another, provided the disk has been correctly read (which I assume is the case given Limine got that far).

The only thing I could think of, which would cause this, is if another volume in the system has a label of `easyos` which is overriding the expected filesystem.

W.r.t. #189, I assume that commit did actually fix the issue? It would be nice to know!

bkauler, 2022-07-13:
I have asked him lots of questions. Also asked him to test 3.10.2.

bkauler, 2022-07-13:
I asked Alfons to try "2:8" instead of "1:8" for the hdd uri, still no go.
He has confirmed the the "easyos" label is unique.

Alfons has sent me a pdf with the tests:

[Overview-works-and-not-works.pdf](https://github.com/limine-bootloader/limine/files/9101817/Overview-works-and-not-works.pdf)

mintsuki, 2022-07-13:
Thanks, that's really helpful. It would also be very helpful to have the full output of `lsvol` on Limine (which may not fit on screen in text mode, in which case please disable `GRAPHICS=no` in the config to have Limine default to a video mode).

bkauler, 2022-07-14:
I have just now forwarded your request to Alfons.

Have received this pdf from him:

[test-the-other-way.pdf](https://github.com/limine-bootloader/limine/files/9106984/test-the-other-way.pdf)

bkauler, 2022-07-14:
Alfons has sent me a photo and a pdf:
[image]

[alfons-test-with-test-drive.pdf](https://github.com/limine-bootloader/limine/files/9112321/alfons-test-with-test-drive.pdf)

mintsuki, 2022-07-14:
It would appear that the internal disk has simply failed Limine's internal MBR validation check. Which is honestly not super surprising considering the disk was formatted long enough ago to have partitions start at sector 63.

One solution would be to run `limine-deploy --force-mbr` on the disk, which will make sure that the MBR is subsequently recognised by Limine.

bkauler, 2022-07-15:
Additional information provided by Alfons:

> I am surprised about the problems encountered!
> 
> Tests with Dell and mbr created by:
> - grub4dos
> - grub1
> - grub2
> - win10
> all unsuccessful, mbr can not be read by limine!!
> 
> - my tests with the Macbook were all successful (partition table gpt)
> - my tests with Dell were only successful with mbr created by limune,  tested with test-drive (see documents: test-the-other-way.pdf, test-with-test-drive.pdf)

mintsuki, 2022-07-15:
Yes, when I am talking about MBR I mean the MBR partition table, ***not*** the MBR code nor bootloader code residing in the MBR.

Installing any bootloader does not change the partition table so it is unsurprising that his testings were all unsuccessful.

The partition table evidently fails to pass Limine's sanity checks, which installing Limine with `--force-mbr` will ensure it will pass in the future, hence why my suggestion.

I could also look into relaxing Limine's MBR sanity checks, which is probably what should be done at this point.

mintsuki, 2022-07-15:
Please try this patch to see if it works with that.
```
diff --git a/common/lib/part.s2.c b/common/lib/part.s2.c
index eadd53e..1055d5c 100644
--- a/common/lib/part.s2.c
+++ b/common/lib/part.s2.c
@@ -247,6 +247,8 @@ struct mbr_entry {
 bool is_valid_mbr(struct volume *volume) {
     // Check if actually valid mbr
     uint16_t hint = 0;
+
+/*
     volume_read(volume, &hint, 218, sizeof(uint16_t));
     if (hint != 0)
         return false;
@@ -254,6 +256,7 @@ bool is_valid_mbr(struct volume *volume) {
     volume_read(volume, &hint, 444, sizeof(uint16_t));
     if (hint != 0 && hint != 0x5a5a)
         return false;
+*/
 
     volume_read(volume, &hint, 510, sizeof(uint16_t));
     if (hint != 0xaa55)
```

bkauler, 2022-07-16:
Alfons has replied with a success story.
Note, he doesn't post here directly, nor to the Puppy Forum, though he reads both. He said that his English skill is so poor, he is too embarrassed to post.
I reckon his English is OK.

Anyway, using me as the go-between, this is what I asked him to do. I wasn't sure from his earlier replies whether he had actually tried the limine-deploy with --force-mbr:

```
Alfons,
mintsuki has provided a patch:

[https://github.com/limine-bootloader/limine/issues/195](https://www.google.com/url?q=https%3A%2F%2Fgithub.com%2Flimine-bootloader%2Flimine%2Fissues%2F195&sa=D&sntz=1&usg=AOvVaw1HouBGf6dEtJJEqmqFGnoS)

I have compiled it, see attached.
Expand the tarball.

# tar -xf limine-3.12-amd64.tar.gz

First, if the Dell drive is /dev/sda, run this, from the usb-stick,
using the 'limine-deploy' from the tarball:

# cd limine-3.12
# ./limine-deploy /dev/sda --force-mbr

If the Dell boots, that is good.
If not copy the file from the tarball 'limine.sys' to the Dell.
Try booting the Dell again.
```
This is his reply:

```
Barry,
yeah, yeah it works!!!
My procedure see attachment.
Regards,
Alfons
```
Attachment below:

[test-limine-3.12.pdf](https://github.com/limine-bootloader/limine/files/9125441/test-limine-3.12.pdf)

Then he sent another email:

```
Barry,
more info:

with this patch i can now also read the existing mbr, not only limine-mbr!
my tests on dell work without making any change to the mbr.
mintsuki statement is correct.

Regards,
Alfons
```
So the patch looks good.

commit ffb4ba6f (2022-07-16): mbr: Relax MBR validation requirements. Closes #195
commit 7e81f5a8 (2022-07-16): mbr: Relax MBR validation requirements. Closes #195
closed 2022-07-16 by mintsuki