Limine-Bootloader/Limine #200: Limine 3.12.2 Linux boot causing reduced detected physical RAM size

nphuracm, 2022-07-22:
**Limine version used**
Limine 3.12.2 (AUR)

**Expected behaviour**
Limine boots Linux (5.18.12.arch1-1) correctly, with all the RAM (15.6 GiB) detected.

**Actual (broken) behaviour**
The detected RAM Size (both in KDE Info Center and the MemTotal section of /proc/meminfo) shrunk to 12.7 GiB/13302272 kB.

**To Reproduce**
On a PC like this:
CPU: AMD A10-7860K
MB: ASUS A68HM-E
RAM: Kingston HyperX Fury 8GB DDR3-1866 x 2 (16GB Dual-Channel)
BOOT: UEFI
0. Install Limine (UEFI), Arch Linux, KDE Plasma, KDE info center and Windows 11.
1. Run the EFI Shell. (Not built-in.)
2. Execute the Linux kernel using EFISTUB, with the initrd path given in the command line
3. Open KDE Info Center. The RAM size shown is 15.6 GiB.
4. Reboot into Limine.
5. Execute the Linux kernel using Limine using the Linux protocol with the initrd path given in MODULE_PATH
6. Open KDE Info Center. The RAM size shown is now 12.7 GiB.
7. Reboot into Limine again.
8. Start Windows 11 by chainloading the bootmgfw.efi file.
9. Open Task Manager. The RAM size shown is 15.9 GB.
**Stacktrace**
None.

**Additional context**
I have plymouth, v86d and uvesafb. Although v86d and uvesafb are still there, configured in my system they never functioned correctly
I have passed `quiet`, `splash` and `rw` to the kernel arguments

mintsuki, 2022-07-22:
Hello.

Did this use to not happen in a previous version of Limine or is it something that always occurred?

Can you please post the `e820` lines from `sudo dmesg` here?

nphuracm, 2022-07-22:
`sudo dmesg | grep e820` output:
`
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000000fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000001000-0x000000000008ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000090000-0x0000000000090fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000091000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000000ffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000001000000-0x00000000bc77efff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bc77f000-0x00000000dd376fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dd377000-0x00000000dd3a6fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dd3a7000-0x00000000dd3c5fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000dd3c6000-0x00000000dd901fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000dd902000-0x00000000de3a5fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000de3a6000-0x00000000de3a6fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000de3a7000-0x00000000de5acfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000de5ad000-0x00000000de9f2fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000de9f3000-0x00000000deff3fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000deff4000-0x00000000deffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feb80000-0x00000000fec01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed8ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000041effffff] usable
[    0.000193] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000197] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000782] e820: update [mem 0xe0000000-0xffffffff] usable ==> reserved
[    0.340539] e820: reserve RAM buffer [mem 0x01000000-0x03ffffff]
[    0.340541] e820: reserve RAM buffer [mem 0xdd377000-0xdfffffff]
[    0.340543] e820: reserve RAM buffer [mem 0xde3a7000-0xdfffffff]
[    0.340544] e820: reserve RAM buffer [mem 0xde9f3000-0xdfffffff]
[    0.340545] e820: reserve RAM buffer [mem 0xdf000000-0xdfffffff]
[    0.340546] e820: reserve RAM buffer [mem 0x41f000000-0x41fffffff]
`

mintsuki, 2022-07-22:
Is this with Limine? If so, post the same information when booted with EFISTUB.

nphuracm, 2022-07-23:
`
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000000fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000001000-0x000000000008ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000090000-0x0000000000090fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000091000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000dd376fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dd377000-0x00000000dd3a6fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dd3a7000-0x00000000dd3c5fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000dd3c6000-0x00000000dd901fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000dd902000-0x00000000de3a5fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000de3a6000-0x00000000de3a6fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000de3a7000-0x00000000de5acfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000de5ad000-0x00000000de9f2fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000de9f3000-0x00000000deff3fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000deff4000-0x00000000deffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feb80000-0x00000000fec01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed8ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000041effffff] usable
[    0.000000] e820: update [mem 0xb8d55018-0xb8d73c57] usable ==> usable
[    0.000000] e820: update [mem 0xb8d55018-0xb8d73c57] usable ==> usable
[    0.000194] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000197] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000784] e820: update [mem 0xe0000000-0xffffffff] usable ==> reserved
[    0.358647] e820: reserve RAM buffer [mem 0xb8d55018-0xbbffffff]
[    0.358650] e820: reserve RAM buffer [mem 0xdd377000-0xdfffffff]
[    0.358651] e820: reserve RAM buffer [mem 0xde3a7000-0xdfffffff]
[    0.358652] e820: reserve RAM buffer [mem 0xde9f3000-0xdfffffff]
[    0.358653] e820: reserve RAM buffer [mem 0xdf000000-0xdfffffff]
[    0.358654] e820: reserve RAM buffer [mem 0x41f000000-0x41fffffff]
`

mintsuki, 2022-07-24:
Can you try 3.12.3 and report back?

nphuracm, 2022-07-25:
I installed the 'limine' AUR package and simply replaced the /boot/EFI/BOOT/BOOTX64.efi file with the new one.
Search results for `e820` in dmesg:
[image]
(This is to avoid layout errors - the messages seem jumbled in my last posts. Sorry.)
KDE Info Center:
[image]
_Translation:_
Processor: 4x AMD A10 ...
RAM: 12.7 GiB RAM
GPU: NVIDIA GeForce GTX 650...
/proc/meminfo:
`
MemTotal:       13286912 kB
MemFree:         9733972 kB
MemAvailable:   11368868 kB
Buffers:          191552 kB
Cached:          1499912 kB
SwapCached:            0 kB
Active:           630580 kB
Inactive:        2236048 kB
Active(anon):       1064 kB
Inactive(anon):  1209500 kB
Active(file):     629516 kB
Inactive(file):  1026548 kB
Unevictable:         512 kB
Mlocked:             512 kB
SwapTotal:             0 kB
SwapFree:              0 kB
Dirty:               152 kB
Writeback:             0 kB
AnonPages:       1136640 kB
Mapped:           717756 kB
Shmem:             35400 kB
KReclaimable:     299096 kB
Slab:             402304 kB
SReclaimable:     299096 kB
SUnreclaim:       103208 kB
KernelStack:        8400 kB
PageTables:        21788 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:     6643456 kB
Committed_AS:    4721272 kB
VmallocTotal:   34359738367 kB
VmallocUsed:       67104 kB
VmallocChunk:          0 kB
Percpu:             2752 kB
HardwareCorrupted:     0 kB
AnonHugePages:    313344 kB
ShmemHugePages:        0 kB
ShmemPmdMapped:        0 kB
FileHugePages:     49152 kB
FilePmdMapped:     43008 kB
CmaTotal:              0 kB
CmaFree:               0 kB
HugePages_Total:       0
HugePages_Free:        0
HugePages_Rsvd:        0
HugePages_Surp:        0
Hugepagesize:       2048 kB
Hugetlb:               0 kB
DirectMap4k:      460076 kB
DirectMap2M:     4784128 kB
DirectMap1G:     9437184 kB
`

mintsuki, 2022-07-25:
Does using the attached image improve the situation? (Please note that it may fail to boot so ensure you have a backup)

https://cdn.discordapp.com/attachments/706151776495206426/1001033617704759356/BOOTX64.EFI

mintsuki, 2022-07-26:
Also test this one if you get the chance, thanks:

https://cdn.discordapp.com/attachments/706151776495206426/1001368667365318757/BOOTX64.EFI

mintsuki, 2022-07-27:
I also implemented the above change in 3.13. Closing this for now. Please reopen if still an issue on 3.13.

closed 2022-07-27 by mintsuki