Limine-Bootloader/Limine #201: BIOS chainload request for clarification

bkauler, 2022-07-25:
Would you kindly clarify something for me. I am confused by this documentation:

> Chainload protocol on BIOS:
> 
>     DRIVE - The 1-based BIOS drive to chainload, if omitted, assume boot drive.
>     PARTITION - The 1-based BIOS partition to chainload, if omitted, chainload drive (MBR).
>     MBR_ID - Optional. If passed, use an MBR ID (32-bit hex value) to identify the volume to chainload.

The problem I am having is the use of those words "BIOS" and "MBR".
A computer with BIOS-firmware may have drives that are either MBR or GPT drives.

The documentation seems to be implying that chainloading a drive in a BIOS-firmware computer will work if it has a MBR, but not if the drive has a GPT.

Possible scenario, there are two drives in the computer, they could both be bootable, but only one would be the default. If Limine is installed to the default one, it should be able to chainload the other.

After all, limine-deploy can be applied to both MBR and GPT drives, so it could have been applied to the non-default drive, also.

So, the question is, how to chainload a GPT drive in a BIOS-firmware computer?
Or, have I missed something in the documentation?

bkauler, 2022-07-25:
Maybe MBR_ID should be more generic, renamed to PT_ID

```
# lsblk --pairs -o PTUUID,PTTYPE --nodeps /dev/nvme1n1
PTUUID="494d4539-ff62-4bb1-b06b-86cc0bcb8b14" PTTYPE="gpt"
# lsblk --pairs -o PTUUID,PTTYPE --nodeps /dev/sdb
PTUUID="c5e0f2d2" PTTYPE="dos"
```

bkauler, 2022-07-26:
Ah, I think that I have answered my own question. Reading-up on GPT structure,  a GPT drive has a "protective MBR", so it is still booting an MBR.
OK, closing this.

bkauler, 2022-07-26:
Ha ha, have to open this up again. Still need some clarification.

Yes a GPT drive has a "protective MBR", and as I showed above, lsblk returned PTUUID for a "dos" drive, which can be used for the MBR_ID parameter.

But what about a GPT drive? Can that PTUUID value for a "gpt" drive be used in MBR_ID?

closed 2022-07-26 by mintsuki

bkauler, 2022-07-27:
I need yet another clarification!

I wrote an overview of BIOS chainloading at my blog:

https://bkhome.org/news/202207/limine-bios-chainloading-enhancements.html

However, I'm not sure if it is entirely correct. If limine.sys is in the same vfat system partition where Windows has its boot files, such as ntldr or bootmgr, does limine-deploy overwrite the Windows VBR?

mintsuki, 2022-07-27:
Nope. `limine-deploy` does not care about the location of limine.sys.

commit 4c6fba82 (2022-07-26): chainload: Add GPT_GUID support and improve documentation. Closes #201
commit ae9ccdde (2022-07-26): chainload: Add GPT_GUID support and improve documentation. Closes #201