Limine-Bootloader/Limine #203: Booting ISO files

zrudyt, 2022-08-03:
I cannot seem to find a way to boot an ISO file in limine.cfg, and according to CONFIG.md, that capability is not explicitly explained.  Does the capability to boot an ISO file exist and it's just the docs that are out of date, or is the capability not there at all?

mintsuki, 2022-08-03:
Define "boot ISO files"?

If by that you mean adding an entry in a Limine config that *literally* boots an actual `.iso` file, then no, that is not a thing, and likely will never be a thing because it's of little use and implementing something close to that requires exotic solutions.

If what you mean is "creating an ISO using Limine as the bootloader", then YES, that is something certainly possible and supported by Limine right now! Just see: https://github.com/limine-bootloader/limine#biosuefi-hybrid-iso-creation

Another thing you could possibly mean is chainloading an ISO burnt onto a USB stick or optical media, that is also totally possible as described in `CONFIG.md`.

closed 2022-08-03 by mintsuki

N1NJ4, 2023-03-11:
I'm also interested (not sure if this is already available under a protocol, limine or chainloading, etc.) Though I thought I'd try clarify the regard in case it made a difference.

An .iso file is effectively like a .zip or folder, contains everything needed for whatever purpose you need. In hence situation like an Windows Install CD, or Linux live cd/iso, they contain the files to boot from, though not accessible as you would from a disk. .iso files are mountable though(at lease within an OS). Hence the availability to "Boot" from an iso, is a very common thing; saving maybe space, though at least file structure and folder layers, although unable to (could though likely not worth noting nor worrying about at this moment in time) write to.. no persistent.

I hope this helped although I'm aware you may already knowledge of such a thing, then I'm interested in the "exotic solutions" (cbb to research though only care if true, lol.). Just doesn't sound like its a whole difference to an actual disk drive; "they are a sector-by-sector copy of the data on an optical disc, stored inside a binary file." layer + layer... etc.

At least hope to see some support to load and boot from bootable .iso's in the future, save's from having to "burn" an iso to a drive(including add the boot data), writing over all contents unnecessarily...