Limine-Bootloader/Limine #209: Limine 3.16 crash on USB 2.0 port

Reynaldo333, 2022-08-23:
Limine 3.16
Fujitsu FUTRO MA574

Has an issue on USB port 2.0. It won`t boot and displays:
> Limine 3.16
> Config file not found
> For information on the format of Limine config entries,consult CONFIG.md in
> the root of the Limine source repository.
> 
> Press a key to enter Limine console...

USB stick will not boot also on USB 3.0 port and unmountable after that.

It has been a week that I am using USB 3.0 on the latest 64-bit Easy OS without any issues.

**For the 32-bit Easy OS, no issues on USB 2.0 and USB 3.0 ports.**

mintsuki, 2022-08-24:
This bug frankly escapes my understanding, assuming it really is a bug to begin with.

Could you try doing the same with the ISO I sent to you on the other issue and seeing if it behaves the same?

Reynaldo333, 2022-08-24:
Yes, it behaves the same.

closed 2022-09-20 by mintsuki

Reynaldo333, 2022-09-22:
Ok...Thanks!

Reynaldo333, 2022-10-11:
This is what I`m getting booting in USB 2.0 port on the test ISO you`ve
sent to me.
[image: DSC_0013.JPG]

after pressing F10 in the editor console...
[image: DSC_0014.JPG]

Best regards,
Reynaldo

Andy-Python-Programmer, 2022-10-11:
@Reynaldo333 The images are not actually attached in the comment you sent above.