Limine-Bootloader/Limine #213: limine 4.0: build fails

Vouivre, 2022-09-08:
I'm the maintainer of `limine` for the kisslinux distribution and the build of version 4.0 fails. 

I'm able to build without problem `limine` 3.21, but `limine` 4.0 fails. Something happened between the two versions. You can get the log file [here](http://0x0.st/of_L.txt). 

If I search for errors in the log file, there is

    clang -cc1as: error: unknown target triple 'aarch64-unknown-unknown-elf', please use -triple or -arch

I build `limine` with
```
./configure \
	--prefix=/usr \
	--enable-all
	
make
make DESTDIR="$1" install
```

where `$1` is a temporary directory created by the package manager. 

My system

```
$ uname -a
Linux HOSTNAME 5.19.6 #7 SMP PREEMPT_DYNAMIC Sat Sep 3 09:48:41 UTC 2022 x86_64 GNU/Linux
```

If you need more information, just tell me what you need.

mintsuki, 2022-09-08:
This is because Limine 4.x adds an aarch64 port which is seemingly unsupported by the Clang/LLVM toolchain you're using to build Limine.

`--enable-all` enables ALL ports. If the UEFI/aarch64 port is not wanted, please pass `--disable-uefi-aarch64` to `./configure` after `--enable-all`.

Vouivre, 2022-09-08:
Yes, I was able to compile with `--disable-uefi-aarch64`. aarch64 is not supported at the moment, so I can disable it. 

Thank you!

closed 2022-09-08 by Vouivre