Limine-Bootloader/Limine #215: protocol: add UART MMIO address tag

Andy-Python-Programmer, 2022-09-10:
The limine protocol should provide UART MMIO address tag; parsing the DTB just for UART is annoying and usually its better to have it before parsing it for better debugging experience.

Itay2805, 2022-09-10:
sounds a it redundant given that the terminal exists, and can provide the same debugging functionality

czapek1337, 2022-09-11:
> sounds a bit redundant given that the terminal exists, and can provide the same debugging functionality in a cross-platform way

You won't be able to easily retain the contents of the terminal after the machine reboots though, which is why serial is superior

Itay2805, 2022-09-11:
Wouldn't it be better to just add support for serial on the terminal tag?

closed 2022-10-30 by mintsuki