Limine-Bootloader/Limine #218: Black screen with WD Elements SE USB3 SSD

bkauler, 2022-09-15:
My daily PC is a Lenovo Ideacentre PC, with Intel i3 CPU. Pretty much stock-standard hardware, and all versions of Limine have booted on it.

I have various OSs installed to the internal drives, with rEFInd boot manager. Up until now, I have only tested Limine when boot from USB-sticks. Today, installed EasyOS, with Limine 3.21, to a WD Elements SE USB3 SATA SSD, and for the first time, boot failure.

Don't even get the menu. Nothing, just a black screen.

I changed to Limine 3.16, same black screen. I changed to rEFInd, and the SSD boots, running from it now.

The only thing that I can think of is that it is a timing problem. The WD SSD is considerably faster than the USB-sticks that I have tested Limine on so far.

mintsuki, 2022-09-15:
Can you please try https://cdn.discordapp.com/attachments/706151776495206426/1019925172955918376/BOOTX64.EFI and see if it is any different?

Andy-Python-Programmer, 2022-09-15:
Nice use of discord there :trollface:

mintsuki, 2022-09-15:
Would be cool if GitHub allowed attaching files, or maybe I'm just too stupid to find out the option to do so.

Andy-Python-Programmer, 2022-09-15:
> Would be cool if GitHub allowed attaching files, or maybe I'm just too stupid to find out the option to do so.
Drag and drop the file.
[image]

mintsuki, 2022-09-15:
That only works for images afaik.

Andy-Python-Programmer, 2022-09-15:
Oh, it requires you to compress it.

mintsuki, 2022-09-15:
Cool, nice arbitrary limitation. Now I know for next time :)

mintsuki, 2022-09-15:
[BOOTX64.EFI.gz](https://github.com/limine-bootloader/limine/files/9574694/BOOTX64.EFI.gz)
@bkauler please try this one as well

bkauler, 2022-09-16:
Black screen with both.

Interesting, I tried legacy boot, and got an error message "limine.sys size incorrect".
I checked, did a "cmp" and limine.sys on the SSD is ok.

This is the SSD I am using, 480GB:

https://www.westerndigital.com/en-au/products/portable-drives/wd-elements-se-usb-3-0-ssd#WDBAYN4800ABK-WESN

I bought it locally for AU$59, my blog report:

https://bkhome.org/news/202205/first-test-wd-elements-se-ssd-480gb-and-comparisons.html

Note, Officeworks no longer stock it. I see on Amazon it is a much higher price.

mintsuki, 2022-09-16:
I honestly doubt the SSD has much to do with the issue.

The incorrect limine.sys size error is very likely caused by an outdated stage 2 installed on the drive. Read: you need to run the `limine-deploy` that matches the `limine.sys` that you have on the disk (or vice versa, you need to copy the `limine.sys` that matches the installed stage 2), else this will happen.

With the second image I sent, I reckon it is pretty unlikely to get a straight up black screen hang. Are you sure the BIOS/firmware is trying to boot off of the disk at all? Maybe the EFI boot entries are somehow malformed? Given that Limine boots on this machine (as we know already) it sounds to me like this may be more likely some configuration error rather than some Limine bug.

That said, I am saying this mainly because I'd like to make sure that that isn't the case before diving deeper into the issue. For now I'll mark this as unconfirmed.

bkauler, 2022-09-17:
There is a simple way to determine if the problem is the SSD or not...

I wrote the latest EasyOS image file, which has Limine 3.21, to a SanDisk Ultra USB-stick.
No problem, it boots, both UEFI and legacy bootup.

Same image file written to the WD SSD, UEFI boot has the black screen. Interesting, this time legacy boot works.

Don't recall, I was messing around trying different things, maybe a replaced limine.sys with a different version before. Anyway, legacy boot works.

For UEFI boot, I replace the 3.21 BOOTX86.EFI with the second one that you posted above, still get a black screen.

Note, the esp boot partition is fat12. That is because the partition is only 7MiB, too small form fat16.

A note about attaching files. I think that you don't have to actually gzip a file, just change its name, with ".gz" appended. That fools the system, at least it works on the Puppy Forum.

mintsuki, 2022-09-17:
[BOOTX64.EFI.gz](https://github.com/limine-bootloader/limine/files/9591027/BOOTX64.EFI.gz)

Can you try this one please? (This time it's a fake .gz)

bkauler, 2022-09-18:
Something different this time, just a little bar on the black screen:

[image]

mintsuki, 2022-09-18:
[BOOTX64.EFI.gz](https://github.com/limine-bootloader/limine/files/9593258/BOOTX64.EFI.gz)
Can you please try this one?

bkauler, 2022-09-18:
Same as the above photo, just that little bar on black screen.

mintsuki, 2022-09-18:
[BOOTX64.EFI.gz](https://github.com/limine-bootloader/limine/files/9593725/BOOTX64.EFI.gz)
Can you try this one? It should print some numbers to determine where the hang occurs.

Andy-Python-Programmer, 2022-09-18:
@bkauler can you remove `QUIET=yes` and insert `VERBOSE=yes` in your limine config.

bkauler, 2022-09-18:
VERBOSE=yes makes no difference, just black screen with the little bar as above photo.

mintsuki, 2022-09-18:
With the latest image I posted?

bkauler, 2022-09-19:
Yes, the latest BOOTX64.EFI

j0534ng31, 2022-09-19:
@bkauler @mintsuki I'm in the same boat...

First of all, I want to thank you both for your work:
- EasyOS has been a perfect option for me: as a computer technician, I've been using it for years at work.
- Limine is a masterpiece and a breath of fresh air for me: It isn't as complex as rEFInd to get started, and the documentation is concise and to the point.

Speaking about the bug, I'm not sure its an exclusive cause of Limine: In my case it's a problem interacting with 'rEFInd Plus' (I'm trying with original 'rEFInd' just now).

Barry, are you using any rEFInd to load/chainload Limine?

Here it works perfectly if I:
- Boot to UEFI shell, run FS0:\EFI\Limine\BOOTX64.EFI, and select EasyOS or any other OS.
- Boot to UEFI shell, run FS0:\EFI\EasyOS\BOOTX64.EFI, and select EasyOS Dunfell.
- Boot to UEFI option Limine (manually created by me), pointing to FS0:\EFI\Limine\BOOTX64.EFI
- Boot to UEFI option EasyOS (manually created by me), pointing to FS0:\EFI\EasyOS\BOOTX64.EFI

No way to make it working (I see the Limine menu and can edit entries, but any boot option ends in a black screen) if I:
- Boot to rEFInd Plus (the one I've tested until now) and select any Limine based option.
  (I have two options right now: The automatic one, and a manual stanza created by me to try to solve or at least narrow down this bug).

I'm going to test it again right now, but if I remember correctly the black screen it's not limited to boot Barry's EasyOS: I also cannot boot other options as chainloading Windows nor booting to UEFI shell.

However, booting Limine directly everything works.

mintsuki, 2022-09-19:
Hello and thank you for reporting the issue.

See, the problem is that a "black screen" isn't really *supposed* to happen. At least not now, and especially not with that last image I posted.

What I did in that last image is to add a series of print statements along control flow, that print using UEFI's own standard output terminal, starting from the very top, at soon as possible after entry. At least in the case of Barry, this should have narrowed down the location of the hang remotely, but apparently it did not, which I find particularily surprising.

What this leaves is a series of educated guesses I can make:
1. Barry did not test the right image. I am certain he did his due diligence and this is not the case, but it's an option.
2. His firmware is the one at fault, hanging itself for whatever reason when trying to load Limine. I also find this to be quite unlikely, especially since earlier on in this issue I have made a change which caused the UEFI terminal's prompt to start showing up on screen (what Barry called the little white bar), meaning Limine got loaded.
3. The debug prints I have added do not show because his UEFI firmware's terminal is broken. It wouldn't be the first time I hear about a machine with a broken UEFI terminal output, but the presence of the prompt makes it also sound unlikely.
4. The `limine-efi` relocation stub hangs, for some reason, instead of either failing or returning to firmware. This shouldn't really happen but as it stands it's sadly the option I am leaning towards, just based on this (limited) remote debugging I could do with Barry.

That's about all the options I could think of at the moment. Though I am sure there may be more that I did not consider.

In your case, I can't tell if it is the same situation as Barry's or not, but we can try debugging that and seeing if it fixes it for Barry as well.

[BOOTX64.EFI.gz](https://github.com/limine-bootloader/limine/files/9598430/BOOTX64.EFI.gz)

Here is an image that should print text very early on and hang, just to sanity check if we enter Limine properly at all or not.

Also, while writing this I realised that one of the reason these test images may have not worked for Barry is that I accidentally left port E9 debug on, which may cause issues on real hardware, but in this last image I have it off, so that also shouldn't be an issue.

bkauler, 2022-09-19:
Now getting some text!
[image]

mintsuki, 2022-09-19:
Hey, very nice!
Now that we're getting somewhere can you try this one out?
[BOOTX64.EFI.gz](https://github.com/limine-bootloader/limine/files/9600466/BOOTX64.EFI.gz)

bkauler, 2022-09-19:
[image]

I wonder if it would be useful to study the source code of refind BOOTX64.EFI, as that works?

mintsuki, 2022-09-19:
> I wonder if it would be useful to study the source code of refind BOOTX64.EFI, as that works?

Nah, I don't think that would be too helpful.

Can you try this new test image? 
[BOOTX64.EFI.gz](https://github.com/limine-bootloader/limine/files/9600902/BOOTX64.EFI.gz)

And thanks for your help debugging this issue, it's much appreciated!

bkauler, 2022-09-19:
No change, same as previous snapshot "12345_"

mintsuki, 2022-09-19:
Can you try this one?
[BOOTX64.EFI.gz](https://github.com/limine-bootloader/limine/files/9601451/BOOTX64.EFI.gz)

mintsuki, 2022-09-19:
[BOOTX64.EFI.gz](https://github.com/limine-bootloader/limine/files/9601467/BOOTX64.EFI.gz)
This one too.

j0534ng31, 2022-09-19:
In my case, the latest two `BOOTX64.EFI.gz` always show **`123456789ab`** indefinitely (so they never get to the limine boot options menu), no matter if booting from the UEFI entry, the auto-discovered rEFInd entry, or my manual rEFInd stanza.

Your `limine v4.1` from GitHub always get to the boot options menu, but only boots EasyOS from the UEFI entries. The two limine via rEFInd entries show four lines of code for a very brief moment, and then the black screen.

This time I have recorded a video, to be able to transcribe the message here:

> linux: Loading kernel \`fslabel://easy/dunfell/vmlinuz\`...
> linux: Boot protocol: 2.15
> linux: Kernel version: 5.15.60 (root@EASYPC15758) \#1 SMP Fri Aug 12 22:
> linux: Loading module \`fslabel://easy/dunfell/initrd\`...
> (a white square cursor)

At this point, I assume that Limine will have finished her task or is about to.

**PS:**
- I'm also using a Lenovo for this test (laptop, i7, 16GB, SSD 1TB, UEFI), and surely with the same UEFI version as Barry...
- But unlike him, I am unable to get this EasyOS version to boot via rEFInd (with or without limine). It only works correctly via Limine when booting directly from the UEFI Shell or from a UEFI entry.

mintsuki, 2022-09-19:
The last 2 images I posted are supposed to hang after the "b", in order to avoid clearing the screen by the menu. So this means that your issue is not quite the same as Barry's.

The fact that you always get the Limine menu and it black screens after the loading messages makes me think that your assessment is most definitely correct with regards to it not being a Limine issue.

bkauler, 2022-09-20:
The second-last one (that I am referring to as "try8") gives this:

123456789ab_

bkauler, 2022-09-20:
The last one ("try9") gives same result.

mintsuki, 2022-09-20:
@bkauler can you please try release 4.1.2 from here? https://raw.githubusercontent.com/limine-bootloader/limine/v4.1.2-binary/BOOTX64.EFI

bkauler, 2022-09-20:
It works!!!

That little bar appeared on the screen briefly, then got the menu, then was able to boot EasyOS.

Is that little bar only there for testing purposes?

Does this success mean that you have recognised what the problem is with my hardware?

And thanks for sticking with it, to solve the problem.

mintsuki, 2022-09-20:
Amazing!

> Is that little bar only there for testing purposes?
That little bar is the UEFI terminal's cursor. Some machines have it, some do not. Think of it as like an MS-DOS text mode blinking cursor, that does not blink.

> Does this success mean that you have recognised what the problem is with my hardware?
It would seem that there was some missing error handling to some firmware calls (that should've been there already, but oh well) and one thing leads to another and you either end up calling the firmware with bogus parameters and it hangs, or you cause some exception, in any case, it's not pretty.

> And thanks for sticking with it, to solve the problem.
Thanks *to you* for sticking through this. As the maintainer it is my duty to take care of bugs like this. Remote debugging is painful, but it sort of is the only way this can be done, as having every users' system and configuration is just not something doable.

closed 2022-09-20 by mintsuki

bkauler, 2022-09-20:
Tried it again, on the USB-stick. This, didn't even get the little bar, just a black screen, 10-sec timeout, then EasyOS booted.

bkauler, 2022-09-20:
Back onto 4.1.2, booting the SanDisk USB-stick. Observing more closely, the little bar is there for the duration, until it times out then EasyOS boots.

Same thing with legacy boot. Get a flashing cursor, 10-sec timeout, then EasyOS boots.

bkauler, 2022-09-20:
I have reconfirmed, 4.1.2 has fixed the problem with the WD SSD. For both UEFI and legacy boot, get the menu.

mintsuki, 2022-09-20:
`QUIET=yes` was set.