Limine-Bootloader/Limine #221: How to debug a NULL response to a request

linkdd, 2022-09-21:
**Limine version used**

4.1.3-binary using BIOS (via QEMU)

**Expected behaviour**

```c
#include <limine.h>

// ...

static volatile struct limine_module_request mod_req = {
  .id = LIMINE_MODULE_REQUEST,
  .revision = 0
};

// ...

void kmain(void) {
  // ...
  assert(mod_req.response != NULL, "should not be null");
  // ...
}
```

**Actual (broken) behaviour**

The response to a module request is `NULL`.

Andy-Python-Programmer, 2022-09-21:
What's your limine configuration.

linkdd, 2022-09-21:
```
DEFAULT_ENTRY=1
TIMEOUT=0

:MyOS
  PROTOCOL=limine
  KERNEL_PATH=boot:///boot/kernel.bin

  MODULE_PATH=boot:///sys/modules/capabilities.bin
```

mintsuki, 2022-09-21:
Your `assert()` macro is either non-standard, or your call to it involves the use of a comma operator which makes the macro call useless. In either case more information is required about your `assert()`.

linkdd, 2022-09-21:
It was just a minimal example, the actual code is:

```c
  if (module_request.response != NULL) {
    printk("Module list:\n");

    for (size_t module_idx = 0; module_idx < module_request.response->module_count; module_idx++) {
      struct limine_file *module = module_request.response->modules[module_idx];

      printk("# %s %s\n", module->path, module->cmdline);
    }
  }
```

Nothing gets printed, hence: `module_request.response` is NULL...

mintsuki, 2022-09-21:
Give a buildable, reproducible example. Also curb your tone.

linkdd, 2022-09-21:
I was only asking if Limine guarantees that this value is never NULL (which means the problem is obviously on my end).

I made a barebones example but it is unable to reproduce the error unless I remove the `MODULE_PATH` from the config file.

I'm closing the issue for now.

closed 2022-09-21 by linkdd

mintsuki, 2022-09-21:
Any response can be NULL if the bootloader does not provide it and should be checked for that. Yes.

W.r.t. your specific issue, I have never seen Limine fail to provide a modules response if a module is properly found and loaded, which leads me to believe this issue is on your end, unless you can produce a reproducible example. If you can do so, feel free to reopen the issue and we'll look into that :)

Andy-Python-Programmer, 2022-09-21:
The module response **can** be NULL if there are no modules requested (this is done to save memory required to allocate the response). In your case, module was specified so, it's an issue on your end.