Limine-Bootloader/Limine #222: Limine 4.1.2 black screen after menu

bkauler, 2022-09-23:
I created a pre-release of EasyOS 4.4, with Limine 4.1.2. Uploaded here:

https://distro.ibiblio.org/easyos/amd64/releases/TEST/dunfell/

The USB-stick boots on my Lenovo PC, the one we have recently de-bugged.

However, I tested on another computer, and got the menu ok, then a black screen. This is limine.cfg:

```
VERBOSE=yes
TIMEOUT=10
GRAPHICS=no
DEFAULT_ENTRY=1
EDITOR_ENABLED=no
INTERFACE_BRANDING=EasyOS Limine Boot Manager

:EasyOS dunfell
    COMMENT=EasyOS Dunfell bootup
    PROTOCOL=linux
    TEXTMODE=yes
    KERNEL_PATH=boot://2/easyos/vmlinuz
    MODULE_PATH=boot://2/easyos/initrd
    KERNEL_CMDLINE=initrd=initrd rw wkg_uuid=88369420-3b1c-11ed-8152-287fcfeb4376 wkg_dir=easyos/
```

This computer has a HP motherboard, and early UEFI firmware. The computer has Limine 3.10.1 installed, to multiboot the internal OSs. Works fine. All prior tests of Limine on this computer have worked.
The latest that I tested, successfully, was a USB-stick with Limine 3.16.

With 4.1.2, there are no error messages after the menu, just goes to a black screen, then unresponsive and need to hold down the power button to power-off.

bkauler, 2022-09-23:
Have tested on another computer, my Mele miniPC, the EasyOS 4.4pre USB-stick boots OK.

mintsuki, 2022-09-23:
Can you check exactly which Limine release stops working on this machine? That would ease my job a lot.

bkauler, 2022-09-23:
OK, testing the 3.x series, working backwards:

3.24 broken, 

3.23 worked first time, but then tested three more times and black screen.
 tried both reboot and start from cold, just get black screen.

3.22.3 strange, also worked the first time, black screen on subsequent boots.

Jumping down a few versions...
3.21 black screen every time.
3.20 black screen.

3.17 black screen, but different this time, get that little bar.
Jumping down to 3.16, yes works. So...

3.16.2 black screen, no cursor.

3.16.1 black screen, that cursor is back.
  Cold start (power off for a minute), it works.
  Reboot, it works.
  Reboot, black screen, no cursor.

Back onto 3.16
  From cold, oh no, now get a black screen!
  Reboot, black screen.
  Tested legacy boot, yep works.

Sometimes working, mostly not, not good.
Posting this, will go back further, see if can find a version that works consistently...

mintsuki, 2022-09-23:
Thank you for testing, it is much appreciated :heart:

bkauler, 2022-09-23:
I found it!
3.12.3 boots consistently.
3.13 gives a black screen, no cursor.

I have to stop now. I it is 3.25am here, need to sleep.

mintsuki, 2022-09-23:
Good night! Thanks for the invaluable info, hopefully by the time you wake up the issue will be sorted out...

bkauler, 2022-09-23:
Turning this computer off right now! But a clarification, 3.13 does display the menu ok, the screen goes black after that.

bkauler, 2022-09-24:
Online again. 
Something very interesting. Regarding issue #218, problem with the WD Elements SE SSD.
Well, 3.12.3 boots on it, except there is a huge delay, black screen for about 35 seconds, before the menu appears. Then the menu works, and bootup success.

mintsuki, 2022-09-24:
[BOOTX64.EFI.gz](https://github.com/limine-bootloader/limine/files/9637563/BOOTX64.EFI.gz)
Can you try this one?

bkauler, 2022-09-24:
Trying it on the PC with old HP motherboard, yes, success!

I rebooted and cold-starts, half a dozen times, to make sure it boots consistently.

bkauler, 2022-09-24:
Also tested it on the WD SSD, yep, boots.

Will this fix be in the 3.x branch?

mintsuki, 2022-09-24:
Yes it will!

mintsuki, 2022-09-24:
Should be upstreamed in 3.25 and 4.3.

I'll close the issue, if it's still a problem feel free to reopen it.

closed 2022-09-24 by mintsuki

bkauler, 2022-09-27:
For anyone who might read this issue, there is some followup debugging in the Puppy Forum:

https://forum.puppylinux.com/viewtopic.php?t=6881

It looks like changing the kernel config from legacy efifb and vesafb to "simple fb" fixes bootup.