Limine-Bootloader/Limine #226: Limine switches to undesired high resolution

bkauler, 2022-09-30:
To explain the problem, I will first show how rEFInd behaves...

refind.conf has this content;

```
timeout 10
textonly on
textmode 1024
showtools shutdown,reboot,firmware
scanfor manual
menuentry "EasyOS Dunfell (in sda6, folder easyos)" {
 volume hdd48gb3
 loader /easyos/vmlinuz
 initrd /easyos/initrd
 ostype Linux
 options "rw wkg_uuid=cef85caf-2d3f-4bbc-9a57-ac5b64fb2fdf wkg_dir=easyos"
}
```
I tested with "textmode 0" and "textmode 1024", same screen:
[image]

The documentation on refind.conf:

https://www.rodsbooks.com/refind/configfile.html

Quoting documentation on "textmode":

_Sets the text-mode video resolution to be used in conjunction with textonly or for the line editor and program-launch screens. This option takes a single-digit code. Mode 0 is guaranteed to be present and should be 80x25. Mode 1 is supposed to be either invalid or 80x50, but some systems use this number for something else. Higher values are system-specific. Mode 1024 is a rEFInd-specific code that means to not set any mode at all; rEFInd instead uses whatever mode was set when it launched._

Now, select from the menu, launches EasyOS and we are seeing text echoed from the 'init' script in the initrd, and a logo:
[image]

From the size of the text, I would guess that the screen is running in 640x480. In graphics mode, otherwise the logo would not display. Here is the code that displays the logo:

```
#190307 display logo, -f = reduces to fit screen
if [ -e /dev/fb0 ];then
 idump -f logo1920x1440.png 2>/dev/null
 [ $? -ne 0 ] && rm -f /dev/fb0 #190313
 #well, the the uefi framebuffer dimensions do not know about widescreen...
 #VxH="$(fbset | grep -o 'geometry .*' | tr -s ' ' | cut -f 2,3 -d ' ')"
 #RATIO=$((${VxH/ */}00/${VxH/* /}))
 #if [ $RATIO -gt 150 ];then #widescreen
 # idump -f logo1920x1080.png
 #else
 # idump -f logo1920x1440.png
 #fi
else
 clear #20220206 no logo, just clear the screen.
fi
```

The wallpaper is 1920x1440, that's why it looks stretched on my widescreen monitor.

640x480 is a good choice, as it is guaranteed to be available on all video cards and monitors. I haven't specified that resolution, rEFInd is just switching to it automatically.

This is the behaviour that I want, as get nice big text when in the initrd. And the wallpaper fills the screen. Note, the kernel video module has not yet loaded, that happens after switch_root.

The next post will describe the behaviour with Limine...

bkauler, 2022-09-30:
Booting with Limine, if I want big text in the menu, "INTERFACE_RESOLUTION=640x480" works great. limine.cfg content:

```
VERBOSE=no
TIMEOUT=10
DEFAULT_ENTRY=1
INTERFACE_BRANDING=EasyOS Limine Boot Manager
INTERFACE_RESOLUTION=640x480

:EasyOS dunfell
    COMMENT=EasyOS Dunfell bootup
    PROTOCOL=linux
    KERNEL_PATH=boot://2/easyos/vmlinuz
    MODULE_PATH=boot://2/easyos/initrd
    KERNEL_CMDLINE=rw wkg_uuid=dac4bdd4-3c0c-11ed-b16c-287fcfeb4376 wkg_dir=easyos/
```
[image]

However, when switch to the initrd, no matter what I try, different versions of Limine, always get this:
[image]

Very odd, the 'idump' utility is seeing a certain screen size, but that is not what it actually is.

So, the question is, why? Why can't I get large text, presumably 640x480 graphics mode, and the wallpaper filling the screen?

bkauler, 2022-09-30:
But I did try RESOLUTION and was ignored, just got the hi-res screen as per above snapshot.

Took another look at it, what I had in an earlier test was "RESOLUTION=800x600x24"
Hmm, ok, took off the "24", now it works!

Perchance the video card doesn't support bit-depth of 32, does Limine test 24? Or lower?

Ok, fixed, but there is still an unanswered question. When it switched to a hi-res screen, as per above snapshot, why does the 'idump' utility not see the correct screen resolution?

With "RESOLUTION=800x600" it works, the wallpaper fills the screen.

mintsuki, 2022-09-30:
That wouldn't appear to be a Limine issue.

mintsuki, 2022-09-30:
When it comes to the resolution question, most graphics cards support 32bpp colour depth. Only really old ones are limited to 24bpp.

In any case, all the video modes supported by a given card are entirely up to the firmware, so what I assume is happening here is that your firmware specifically does not support 800x600x24bpp.

When Limine fails to set the requested resolution, it tries to detect it automatically. *It setting the video mode to native resolution is a feature, not a bug!*

Also worth noting that for the interface terminal itself, one may use the font scaling feature as described in the documentation to achieve bigger text without sacrificing actual screen resolution; though this is not something that can be done for the Linux console, only for the Limine terminal itself.