Limine-Bootloader/Limine #230: wtf happened?!

Just-Xezon, 2022-10-07:
none of my projects that uses limine does work!

[image]

lukflug, 2022-10-07:
quality issue

Just-Xezon, 2022-10-07:
quality issue?

mintsuki, 2022-10-07:
I think this is what we call, in technical jargon, an issue of skill

closed 2022-10-07 by mintsuki

Dennisbonke, 2022-10-07:
Seems to be an old version and you're providing exactly zero information for us to determine what is going on. Could you please provide relevant information, like version information and such?

Just-Xezon, 2022-10-07:
mkdir -p iso
	cp $(KERNEL_TARGET) limine/limine.sys limine/limine-cd.bin iso/
	echo 'TIMEOUT=3' 						> iso/limine.cfg
	echo ''                					>> iso/limine.cfg
	echo ':$(OS_NAME)'           		    >> iso/limine.cfg
	echo ''                					>> iso/limine.cfg
	echo 'PROTOCOL=multiboot2'			>> iso/limine.cfg
	echo 'CMDLINE=hello'					>> iso/limine.cfg
	echo 'KERNEL_PATH=boot:///kernel.bin'	>> iso/limine.cfg
	xorriso -as mkisofs -b limine-cd.bin -no-emul-boot -boot-load-size 4 -boot-info-table --protective-msdos-label iso -o $(IMAGE_TARGET)

lukflug, 2022-10-07:
"quality issue" as in the quality of this GitHub issue

Just-Xezon, 2022-10-07:
i have tryed every version it used to work on like 3.0, currently trying to get it to work on v4.0

Dennisbonke, 2022-10-07:
> i have tryed every version it used to work on like 3.0, currently trying to get it to work on v4.0

You got an exact number of that version tho?

Just-Xezon, 2022-10-07:
v3.0-binary and v4.20221007.0-binary

Dennisbonke, 2022-10-07:
The screenshot is from v4.20221007.0?

Just-Xezon, 2022-10-07:
no v3.0, i get the same on v4 to

lukflug, 2022-10-07:
dubious

Dennisbonke, 2022-10-07:
> no v3.0, i get the same on v4 to

Doubt, autodetection isn't in v4.

Just-Xezon, 2022-10-07:
kinda weird it used to work last week no problem

Dennisbonke, 2022-10-07:
Was last week on v3.0 by any chance?

Dennisbonke, 2022-10-07:
Also can you be more specific about the 3.0 version?

Just-Xezon, 2022-10-07:
its just v0.3-binary in branch

Dennisbonke, 2022-10-07:
Latest commit in that branch?

Just-Xezon, 2022-10-07:
yes

Dennisbonke, 2022-10-07:
X to doubt, as the latest 3.0 binary doesn't have autodetection

lukflug, 2022-10-07:
x

Just-Xezon, 2022-10-07:
i dont know i just auto download limine v3.0-binary

git clone https://github.com/limine-bootloader/limine -b v3.0-binary --depth=1
make -c limine

Dennisbonke, 2022-10-07:
yeah that's not the branch, you're using 3.0 exactly which is old as hell. Newer versions don't do autodetect anymore. Issue of skill, update to newer versions.

Just-Xezon, 2022-10-07:
i fucking tryed and it does not work!

Dennisbonke, 2022-10-07:
Yeah but you gave 0 information as to why it doesn't work. Even your screenshot is outdated as fuck. Thanks for playing, try again later.

lukflug, 2022-10-07:
v3.0-branch-binary?

Andy-Python-Programmer, 2022-10-07:
> i dont know i just auto download limine v3.0-binary
> 
> git clone https://github.com/limine-bootloader/limine -b v3.0-binary --depth=1
> make -c limine

```
cd limine
git log
```