Limine-Bootloader/Limine #232: unpacked kernel file size

CorruptedByCPU, 2022-10-29:
Hey!

Its good to know where is the "unpacked" kernel from ELF file, but it would be nice to know also what size (Bytes) of space it takes, wouldn't it?

_*** Nice to have. ***_

[image]

czapek1337, 2022-10-29:
You should map code and data with different permissions, I think adding a `size` field to that would only discourage that since you could map everything with same permissions instead. Just place some symbols around .text, .data and .rodata sections and map everything with the right permissions.

CorruptedByCPU, 2022-10-29:
That's a good point.

closed 2022-10-29 by CorruptedByCPU