Limine-Bootloader/Limine #233: Blank screen on multiple graphic card system

vejnar, 2022-11-10:
I tested successfully Limine v4.20221102.0 on two desktop computers. One with integrated AMD, one with integrated Intel GPUs. Everything works nicely.  It's very convenient. Thanks for the good work!

I am using this config (for the AMD):
```
TIMEOUT=20
TERM_FONT_SCALE=2x2

:Arch Linux
    PROTOCOL=linux
    KERNEL_PATH=boot:///boot/vmlinuz-linux
    CMDLINE=root=LABEL=archroot rw rootfstype=ext4 fbcon=map:1
    MODULE_PATH=boot:///boot/amd-ucode.img
    MODULE_PATH=boot:///boot/initramfs-linux.img
```

I wanted to also use Limine on one of our Supermicro server. This server has an VGA output (independent from the processor) and I added an AMD GPU. The BIOS/UEFI selects automatically the output based on the cable present. It works (ie proper output is used) for the UEFI/BIOS config interface, the boot device menu etc. And importantly also for UEFI console and systemd-boot. However:

* When I plug my monitor to the AMD GPU, and it comes to display Limine, the monitor stays on but nothing is displayed. When the timeout is done, then the system boots normally.
* When I plug my monitor to the integrated VGA output, everything works normally - BIOS/UEFI interface, Limine and system.

I tried to add the option `INTERFACE_RESOLUTION=800x600` (and remove `TERM_FONT_SCALE`) but it didn't change anything.

Since everything before and after Limine seems to properly display on the active output, I don't understand why Limine doesn't do the same. It seems it only works when I use the first output (ie VGA). This is quite inconvenient as I always have a monitor connected on the AMD GPU not the VGA.

I can test versions of Limine with a fix if you can't reproduce the problem.

mintsuki, 2022-11-10:
Hello! Thanks for reporting the issue.

Since I'm not sure what the issue is exactly, I hope you won't mind helping me figure it out by testing a couple of builds.

I'd start with the attached one. Just simply remove the fake .gz extension needed for GitHub to swallow the upload.

[BOOTX64.EFI.gz](https://github.com/limine-bootloader/limine/files/9977369/BOOTX64.EFI.gz)

vejnar, 2022-11-10:
Thanks for getting back to me so fast.

mintsuki, 2022-11-10:
In that case, try this one too, and report back what happens with both images. Thanks for your time.

[BOOTX64.EFI.gz](https://github.com/limine-bootloader/limine/files/9978224/BOOTX64.EFI.gz)

vejnar, 2022-11-10:
Thanks. I tested both. The first one showed no improvement but good news: the second one worked!

Since they have the same filename and size, to identify them:
```
6c02332a0d83240d62245233b4d33bd2  BOOTX64.EFI
e85bef1244c1d60d6681c2c73ad82505  BOOTX64.EFI
```
The e85... worked.

mintsuki, 2022-11-12:
Okay, so the way I addressed this issue (namely the firmware providing multiple framebuffers), at least for now, is by adding a `GOP_INDEX` config parameter.

Setting the parameter to `GOP_INDEX=1` should work in your case. Let me know if it doesn't.

This change has been added since 4.20221112.0.

closed 2022-11-12 by mintsuki

vejnar, 2022-11-12:
Thanks for the fix and the new release.

I would consider this a temporary fix though: manually setting output isn't convenient at all. It's easy to imagine scenario where changing that parameter can prove to be difficult (dead GPU). Also, other bootloader (systemd-boot for instance) are able to detect the GPU used. Since the BIOS/UEFI is doing output selection, how come that Limine can't get that info?

mintsuki, 2022-11-12:
Good question!

Limine, unlike other bootloaders, uses its own terminal which draws directly onto *a* framebuffer. I have tried searching in the UEFI specification if there is a way to retrieve the primary framebuffer used by `ConOut` to no avail.

If that is not possible, the alternative, besides what we have now, would be to fallback onto the EFI console (which Limine can already do) whenever multiple framebuffers are present.

It isn't a pretty solution but it should work. I will further inspect the UEFI specification to determine whether finding the primary framebuffer is at all possible, else I will integrate this fallback instead.

vejnar, 2022-11-12:
Thanks for the explanation.

The more robust is a bootloader the better it is in my opinion, so fallback option are not necessarily bad. I think your plan makes sense.

You are mentioning that Limine can already fallback to EFI console, is that already a configurable option?

mintsuki, 2022-11-12:
Nope, but I will make it so alongside the aforementioned changes.

Itay2805, 2022-11-12:
From what I am seeing there are two vars named `ConOut` and `ConOutDev`, looking at the uefi spec they contain the default, and the possible output devices 
[image]
[image]

Maybe trying to get the `EFI_DEVICE_PATH_PROTOCOL` directly from the ConOut will just work, but if not it is def present in a UEFI var that you can query and then search for the GOP device based on that device path.

Parsing mine from linux yields:
```
PciRoot(0x0)/Pci(01|00)/Pci(00|00)/AcpiAdr(0x80010370)
```
which seem to mostly align with my PC where the GPU is on `01:00.0`, now I am not sure what is up with the rest of that but I think it might not be important if all we care about is finding the GOP based on device path (worst case we can probably find some examples in OVMF).

mintsuki, 2022-11-12:
Hello @vejnar 
Can you try this image and report back?
[BOOTX64.EFI.gz](https://github.com/limine-bootloader/limine/files/9995705/BOOTX64.EFI.gz)

mintsuki, 2022-11-12:
Try this one as well
[BOOTX64.EFI.gz](https://github.com/limine-bootloader/limine/files/9995783/BOOTX64.EFI.gz)

vejnar, 2022-11-14:
I tested the released 20221112 adding GOP_INDEX to my config. It doesn't work (blank screen). I tested with GOP_INDEX=0, 1 and 2 with the same result.

For the images from 2 days ago:
```
f6337ec5769a8d61c68c8f9a1b52dcae  BOOTX64.EFI
ddde9fcd74ba6b1c1fae619bbd61bbbd  BOOTX64.EFI
```
I tested both, without GOP_INDEX. Neither of them worked.

I reverted back to the e85... version. This is the only one working so far.

mintsuki, 2022-11-14:
Try booting the latest version with `SERIAL=yes` in the config.

vejnar, 2022-11-28:
I removed GOP_INDEX and tried version `v4.20221119.0`. I still got a blank screen.

Adding `SERIAL=yes` fixed the problem.

Although the interface is textual (and less pretty). I think this is fixing this bug but this is a workaround and not fixing the more generic problem of automatic output detection. Also the e85... version is still working. Maybe that will provide some more insights to you on how to address the problem.

mintsuki, 2022-11-29:
The reason the famous e85 commit works is that I ran a test to see if the issue was related to different GOP outputs, where I simply hardcoded outputting to the 2nd GOP. That is obviously not a good fix since it would break 99.9% of Limine installs where the machine does not use multiple GOPs.

If you're feeling brave, I've actually been working on a more proper fix for a while now, part of which is upstreamed in trunk. So if you wanna give that a shot and report back if it works, that'd be nice!

mintsuki, 2023-02-16:
@vejnar Finally, I merged multiple framebuffer support in Limine 4.20230216.0!

Let me know if it works if you can, thanks!

vejnar, 2023-03-14:
I tested v4.20230314.0 and it works!

Only weird thing is that Limine is using only about a third of the top left surface of the monitor (while the BIOS before and systemd after Limine are using the full surface). I tried adding `TERM_FONT_SCALE=2x2` in the config but it increasing the font size but not the surface used. I could only test it on one monitor.

Thanks @mintsuki

mintsuki, 2023-03-14:
Great to hear!

The "issue" you're describing is more than likely intended behaviour, if what you mean is the menu box being limited to a top left corner with a smaller size than what is possible.

This is because in order for Limine's interface to handle multiple GOPs, I made it broadcast the console to every screen, and in order for things to be correctly mirrored, it picks the rows and cols size from the smallest GOP it found.

Evidently the smallest GOP it found is the one of the screen that you're not looking at (it is probably not even shown anywhere as per above conversation, but it does exist), hence why the limited size of the menu shown.