Limine-Bootloader/Limine #235: Headless mode

Vouivre, 2022-11-14:
Hi,

first feel free to close the issue if you only accept bug reports. I have a question the only way I found to ask it is to create an issue. 

I use `limine` to boot a computer without a screen. When I boot with a monitor plugged into my computer, it boots fine. Without a monitor, so in headless mode, I suspect there is a problem. I that case, I suspect there is problem, because I can't find my computer in my local network after starting my computer.

I suspect the problem hast to do with the bootloader. There is another similar issue [here](https://bbs.archlinux.org/viewtopic.php?id=233260). In order to work with `grub`, I have to pass the parameters `nomodeset text` to the kernel and in `grub` it's necessary to set the option `GRUB_TERMINAL_OUTPUT=console`. 

To pass a parameter in the kernel, there is no problem:

```
CMDLINE=root=/dev/sda1 ro loglevel=3 quiet nomodeset text
```

I suspect, in order to work I have to set the output to console and disable all graphics, but I don't know how. I would like to force the boot menu in text mode, but `GRAPHICS=no` doesn't work because I'm on UEFI. I have tried `TEXTMODE=yes` but it didn't help. 

On a UEFI system, how can I:

- set the boot menu in text mode
- set the output of limine only in text mode.... maybe it has nothing to do with limine, but with the kernel.

mintsuki, 2022-11-14:
Hi,
Text mode is not a thing on UEFI systems, so it can't be set.

From my testing, Limine is perfectly capable to boot a kernel in headless mode without any special config settings; so I'd first rule out that the issue isn't on your end, either kernel or user side, first.

closed 2022-11-14 by mintsuki

Vouivre, 2022-11-14:
Thank you for the information. So I can exclude any configuration missing in limine.