Limine-Bootloader/Limine #237: README confusing about filesystems

stacksmith, 2022-11-19:
`Even though these are the filesystems Limine supports directly, any filesystem can be utilised by the kernel provided that the kernel (and any needed modules) are on a directly supported filesystem.`

I think what you are saying is that the kernel must reside on a supported filesystem and may implement and use any other filesystem types(or anything else for that matter).

But I am not sure.

Maybe it is best to remove the paragraph as it really says absolutely nothing.  If the reader is not aware that they can write a kernel and filesystem drivers, this paragraph is not helpful.  Your target audience should know this anyway.

czapek1337, 2022-11-19:
Yeah it is kind of confusing for someone who's not very technical, I suggest we change this paragraph to "Limine can load it's configuration, the kernel and modules from any of the filesystems listed above" or something alike.

stacksmith, 2022-11-19:
That is much better!

closed 2022-11-30 by mintsuki