Limine-Bootloader/Limine #244: Feature Request: aarch64 Linux protocol

yveszoundi, 2023-01-09:
I distribute a Live CD as zero configuration option for [one of my GitHub projects](https://github.com/rimerosolutions/entrusted). I've been evaluating Limine for its quality, appearance, simplicity and ability to produce painlessly small hybrid ISO images.

### Request

This is a feature request for implementing the `Linux` protocol for `aarch64`. I initially didn't read much of the documentation and I then realized at boot time that this is a `TODO` item for `aarch64`.

- It appears that the only available protocol for my use-case is `Limine`, however I'm not building "my own kernel"
- I tried extracting an ELF image from a `vmlinuz` ([vmlinuz-to-elf](https://github.com/marin-m/vmlinux-to-elf)), instead of reading more the docs...

### Rationale

For many users and/or Live CD distributors, I think that Limine can easily become the de-facto boot manager option with `aarch64` Linux protocol support.

- For an hybrid x86_64 ISO (BIOS and UEFI), Limine came as a winner for the image size (~670 MB) , followed by `Syslinux` and `Systemd-boot`  (~715 MB) and then `Syslinux` and `Grub` (~745 MB).
- I didn't need to read/understand much, prior just trying it in my Live CD build pipeline

qookei, 2023-01-10:
> It appears that the only available protocol for my use-case is `Limine`, however I'm not building "my own kernel"

Limine also supports chainloading EFI executables, which means if your kernel is compiled with EFISTUB support (`CONFIG_EFI_STUB=y`), you can chainload it from Limine, although it currently does not seem to support specifying the kernel command line (I think it should be possible to implement though).

That being said I should get around to implementing the Linux boot protocol support one day or another. Though using it comes with the caveat of not working on every machine, as it requires a device tree, and UEFI may not provide it, although there possibly is a setup option to enable it (e.g. on the RPi4), and it could be worked around by loading it from a file (but that would make the image specific to a certain SoC).

yveszoundi, 2023-01-11:
Thank you for the prompt and clear response. I'll see if I can get away with an EFI stub.

mintsuki, 2023-01-17:
As of ac1c5d1 (latest release as well) Limine supports specifying command lines for EFI chainloading. Closing this issue for now.

closed 2023-01-17 by mintsuki