Limine-Bootloader/Limine #245: Failed to apply relocations

calwe, 2023-01-11:
**Limine version used**
[v4.20221230.0](https://github.com/limine-bootloader/limine/commit/83e6e5b4da7791ade3b367ceec8ce82c469d06db)
UEFI

**Expected behaviour**
I expect that my kernel should load error free.

**Actual (broken) behaviour**
On attempting to launch my kernel from limine, I am greeted with the follow error:
```
elf: Unknown RELA type: 0x7
PANIC: elf: Failed to apply relocations
```
This only occurs when I am referencing a symbol (function) defined within my rust code from an external assembly file linked in using the global_asm!() macro. From my understanding this should be all thats required to call a function.

**To Reproduce**
1. Create a function within a rust kernel, marked `extern "C"` aswell as `#[no_mangle]`
```rust
#[no_mangle]
pub extern "C" fn _within_rust {
    // ...
}
```
2. Create an external asm file with a function that pushes the address of the external rust function onto the stack.
```gas
.extern _within_rust
.global _push_onto_stack
_push_onto_stack:
    push _within_rust
    ret
```
3. Include this asm file into the rust program with `global_asm!(include_str!("asm.S"))`
4. Call the function
```rust
unsafe { _push_onto_stack(); }
```

**Stacktrace**
Stacktrace:
  [0xc2da] <panic+0x76>
  [0x1d59c] <elf64_load+0x71c>
  [0x19d7d] <limine_load+0x10d>
  [0x13102] <boot+0xe2>
  [0x12d60] <_menu+0xbb0>

**Additional context**
Calling assembly within rust is not a problem, and simply calling `_within_rust` from the assembly file also does not error, it is only when pushing this symbol onto the stack or moving into a register that the panic occurs.
I'm using the linker file from the barebones limine-rs project without modifications, although I am unsure whether this is a linking problem.

xvanc, 2023-01-12:
You need to get the address of the function using an RIP-relative operation first, to avoid getting that type of relocation.
This should work:
```asm
    lea  rax, [rip + _within_rust]
    push rax
```

calwe, 2023-01-12:
This worked, thanks! Apologies, my assembly knowledge is quite limited :)

closed 2023-01-12 by calwe