Limine-Bootloader/Limine #247: Unable to use fslabel to chainload Windows bootloader

vejnar, 2023-02-06:
* I have a system with 2 drives: one linux and one windows. Each of them has an EFI partition (with different labels).
* The `limine.cfg` is on the linux drive (ext4).
* Linux boots normally using:
```
:Arch Linux
    PROTOCOL=linux
    KERNEL_PATH=boot:///boot/vmlinuz-linux
    CMDLINE=root=LABEL=archroot rw rootfstype=ext4
    MODULE_PATH=boot:///boot/intel-ucode.img
    MODULE_PATH=boot:///boot/initramfs-linux.img
```
* Windows boots normally if I use **guid**:
```
:Windows
    PROTOCOL=chainload
    IMAGE_PATH=guid://5785b197-8a7d-42ee-93df-65df21c5c19c/EFI/Microsoft/Boot/bootmgfw.efi
```

I would like to have a generic config file, so I tried to boot using **fslabel**:
```
:Windows
    PROTOCOL=chainload
    IMAGE_PATH=fslabel://EFI system partition/EFI/Microsoft/Boot/bootmgfw.efi
```
But Limine panics as it can't find the efi file.

I tried to list the partitions in the Limine console to see if the label is properly detected but I have too many partitions and since the console doesn't have pagination I can't see the EFI partition for Windows.

Using `blkid` on linux, I can see the EFI windows partition, with its label and UUID:
/dev/nvme1n1p2: UUID="5688-DDE6" BLOCK_SIZE="512" TYPE="vfat" PARTLABEL="EFI system partition" PARTUUID="5785b197-8a7d-42ee-93df-65df21c5c19c"

I am doing something wrong? Please help.

mintsuki, 2023-02-06:
Hi. The `fslabel` mechanism utilises *file system* labels, not GPT labels (what Linux is referring to here as a "PARTLABEL").

closed 2023-02-06 by mintsuki

vejnar, 2023-02-08:
Thanks for explanation. Indeed windows ESP label is only set in the GPT, with the VFAT partition label left empty.

To fix it, I set the filesystem label using:
```
fatlabel /dev/nvme1n1p2 ESPWIN
```

And then configured Limine using:
```
:Windows
    PROTOCOL=chainload
    IMAGE_PATH=fslabel://ESPWIN/EFI/Microsoft/Boot/bootmgfw.efi
```
Then windows boots normally.