Limine-Bootloader/Limine #256: Unable to chainload unified kernel images

NymanMatthias, 2023-03-12:
I wanted to try using Limine as a bootloader on void linux to boot into a unified kernel image. I assumed that I would need to use the chainload protocol, which led my `limine.cfg` to look like:
```
TIMEOUT=5

:Void Linux
    PROTOCOL=chainload
    IMAGE_PATH=boot:///EFI/Linux/6.2.3
```
However, when I try to boot into them, they fail with `unable to locate embedded .linux section: not found`. However, when I try to boot directly into the same unified kernel image from UEFI, this error does not appear.
My unified kernel images are generated as per the [Arch Wiki](https://wiki.archlinux.org/title/Unified_kernel_image#Manually), and my efistub comes from gummiboot.
Is this an error on my part?

My limine version is 4.20230308.0

mintsuki, 2023-03-12:
I have no idea. Mind sending the image you're trying to load over, so that I can test it myself?

NymanMatthias, 2023-03-12:
Github doesn't allow uploading EFI binaries, where should I send it?

mintsuki, 2023-03-12:
I think GitHub allows you to upload any file as long as it ends in `.gz`, regardless of whether it's actually gzipped or not.

Alternatively use gdrive or whatever you find easy.

NymanMatthias, 2023-03-12:
Here is the file, compressed with gzip. This is not the exact same file, as the other kernel had a host specific config, and was unlikely to boot elsewhere, some cmdline arguments have also been removed.
[6.1.15_1.efi.tar.gz](https://github.com/limine-bootloader/limine/files/10951649/6.1.15_1.efi.tar.gz)

mintsuki, 2023-03-12:
That image does not appear to be a valid UEFI executable nor a valid Linux kernel file.

NymanMatthias, 2023-03-12:
Hmm, weird. Here is a the same kernel, with a smaller initramfs, so I could just rename it rather than compressing it.
[6.1.15_1.efi.gz](https://github.com/limine-bootloader/limine/files/10951719/6.1.15_1.efi.gz)

mintsuki, 2023-03-12:
This one works.

I tried to reproduce the issue in QEMU, but failed to do so. Chainloading this image works fine, and I've been able to get output from the kernel out of serial by using command line options.

Perhaps this message thread is relevant to the issue: https://www.spinics.net/lists/systemd-devel/msg08272.html ?

NymanMatthias, 2023-03-12:
It seems the issue wasn't with my config or limine, but rather gummiboot. I intended to do some testing with systemd-boot's efistub rather than gummiboot's, and didn't notice that, the image i sent you was built with systemd-boot's efistub rather than gummiboot's.

I can now sucessfully boot, as long as the efistub provider is systemd-boot.

closed 2023-03-12 by NymanMatthias